cetic/6lbr #243: Interfacing 6LBR with a Simulated WSN (COOJA)

MrMikeyMouse, 2017-01-16:
Hi,

I'm running example. Everything is good because i can ping addresses from cooja but when i want to set ports to enjoy NAT64 i see this:

[image]

In my noode-config i have this:

[image]

What should i do?

laurentderu, 2017-01-16:
Do you see parsing errors in 6lbr.log ? I believe you can not put on the same line the header for the node [...] and the parameter configuration name=...

MrMikeyMouse, 2017-01-16:
It's just example, i don't see logs, hmmm. But many thanks I've corrected and now i have
[image]

ok i have NAT64 on 192.168.52.2 and i my eth0  is set on 192.168.52.128 and i can ping my NAT64 but when i type [192.168.52.2:25000] and it's not working(i run http on my nodes it's working when i type http://[aaaa::202:2:2:2]/ i see "websense"] What i doing wrong?? :(

laurentderu, 2017-01-18:
If NAT64 is configured properly and your Eth interface has a valid IPv4 configuration and your node has a HTTP server enabled, it should work without problem. Could you set the log level to 5 and see if you try to access your node that you have outgoing packets on the WSN ?

closed 2017-06-15 by laurentderu