cetic/6lbr #244: NVM corruption in CC1310 Launchpad+ ENC28j60

marciolm, 2017-01-19:
Hello,
I am running sucessfully the 6LBR Version : 1.5.x (Contiki-develop-20161222) in a cc1310 Launchpad + ENC28j60 ethernet adapter. 
Just after the firmware upload, the 6LBR works perfectly, but if I do a reset in the board, it stops to work, and back only after a reflashing.
I guess is a NVM corruption due a bug, because when I commented out the lines calling the function "nvm_data_write()" in  the 6lbr/nvm-config.c file, the problem ceased.  I'm not sure if there is a need to write the NVM anymore, because the  6LBR "lite" version unfortunatelly don't have the runtime configuration tabs.

Regards
Márcio

marciolm, 2017-01-19:
Solved after changing the include file from "cc26xx-int-nvm.c" to "cc26xx-ext-nvm.c" in  /6lbr/examples/6lbr/platform/srf06-cc26xx/Makefile.srf06-cc26xx
Thanks
Márcio

closed 2017-01-19 by marciolm

laurentderu, 2017-01-20:
Thanks for the feedback ! I will add a compilation flag to make this easier and document this in the Wiki.