cetic/6lbr #245: [Question] Uninstall 6lbr

masb3, 2017-01-20:
1) How to uninstall 6lbr?
There is no RPi 1.4.0 Package so I built it from source with 'make all'.
2) In tutorial is said to configure 6lbr.conf , but I don't see such a file, should I modify 6lbr.conf.rpi?
3) Are there any specific configurations in case I connect my RPi 3B wirelessly?

laurentderu, 2017-01-20:
We removed the binary packages because they were not compatible with all RPi/BeagleBone. Once the correct applicability is found they will be back.

You can use 6lbr.conf.rpi as a base to create your 6lbr.conf, or use 6lbr.conf.example which contains all the parameters and their documentation.

For using the WiFi, please check this thread : https://github.com/cetic/6lbr/issues/237

masb3, 2017-01-20:
@laurentderu Thank you for reply. After checked thread #237  I decided to not use WiFi, instead use Ethernet. I disabled Wifi, connected eth0 and started 6lbr with sudo service 6lbr start. Unfortunately nothing happend, wanted to check 6LBR address in /var/log/6lbr.ip but that file not existing. In /var/log/6lbr.log - "Waiting for /dev/ttyACM0". The name of my ethernet connection is eth0. My /etc/6lbr/6lbr.conf is
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=1

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only

PS. I want to use cc2650 launchpad as slip-radio and end nodes, but they are on the way to me, is it a reason why 6lbr is not started?

chenek, 2017-04-27:
It seems that there is something wrong with your slip radio. Do you connect slip-radio as ttyACM0?

closed 2017-06-15 by laurentderu