cetic/6lbr #246: Last Update 6lbr on Contiki

MarCast, 2017-01-26:
Until recently I could connect to the 6lbr link [bbbb :: 100] without any problem. Today I tried to redo the system from scratch and can not connect on the sensors link. It may be that the cause is the new update?

aparvizi, 2017-01-27:
i have this problem also.

with [bbbb::100], display index page, but other pages not display , i think web services crashed.

MarCast, 2017-01-27:
@g-oikonomou  Can you help us?

aparvizi, 2017-01-27:
build this branch, problem will be solved:

git clone https://github.com/cetic/6lbr -b develop-20170121

MarCast, 2017-01-27:
How? You've tried? It works?

MarCast, 2017-01-27:
@aparvizi have you try? Now your system is ok?

MarCast, 2017-01-27:
@laurentderu Can You help me?

aparvizi, 2017-01-27:
yes, its ok

MarCast, 2017-01-27:
One question! I can install all from zero (on raspberry) or can i do git clone https://github.com/cetic/6lbr -b develop-20170121 now?

aparvizi, 2017-01-27:
i test on RPI 2, 
see:
https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration

but all step as below:
git clone https://github.com/cetic/6lbr -b develop-20170121
cd 6lbr
git submodule update --init --recursive
cd examples/6lbr
git checkout develop-20170121 

BUILD:

make all #all_native for version < 1.4
make plugins
make tools

INSTALATION:
make install
make plugins-install
update-rc.d 6lbr defaults

MarCast, 2017-01-30:
Ok, I'll try and let you know

MarCast, 2017-01-30:
Ok! It works. Thanks you!

electronut, 2017-03-06:
Thanks very much for this post. I had the same issue - the `[bbbb:100]` webpage would crash after the first load. Solved by switching to `20170121` version. @laurentderu when might this change get merged with the master branch?

laurentderu, 2017-03-08:
@electronut No definite date yet, but I hope before end of this month, at least for a release candidate for the 1.5.x stream and a maintenance release for 1.4.x stream.

closed 2017-03-08 by laurentderu