cetic/6lbr #247: 6LR mode

ronnie1233, 2017-01-30:
Hello guys i am working on a mesh topology network . Now i am confident  with CC1310 nodes are able to display in 6lbr mode . 

I  now want my each intermediate node to act as router in radio level which acts 6lowpan router with in 6lbr.
I am currently working on RFC 6775  This is the link : https://tools.ietf.org/html/rfc6775

So how do you actually use a 6lr  with CC1310 which acts Router c0ordinator with other  leaf nodes . 

I need help regarding this topic !! If i missed  the documentation in Cetic itself i need little regarding this .

laurentderu, 2017-02-01:
Currently in Contiki all the nodes act either as RPL Router or RPL Leaf, there is no implementation of RFC  6775. There is a branch of 6LBR with RFC 6775 support integrated, but it's rather outdated and does not support CC1310 I'm afraid.

ronnie1233, 2017-02-02:
Thanks @laurentderu  is there any updated work going on this part .