cetic/6lbr #248: custom client to communicate with 6lbr

jvasanthan, 2017-01-30:
Hi to all,

Thanks to 6lbr community for developing such an awesome project

I am new to 6lowpan and 6lbr sorry for newbie mistakes!!

I am working on 6lopwan with the following goals in mind
1. multiple CC2650 nodes running contiki
2. run 6lbr in rpi3 but I dont want to run a webserver rather a custom client

basically the data from WSN will be received in RPi3 and we will be running a custom client application to communicate with our custom server.

How can I make my custom client to communicate with 6lbr to access data from WSN?

I appreciate your inputs

James

ronnie1233, 2017-02-01:
Use udp clients or web demo examples from contiki and run it on your launchpad

jvasanthan, 2017-02-01:
Thanks ronnie1233

I will run udp client in Contiki (the contiki example for CC2650 has examples). How to communicate with this udp client from 6lbr. Where should I modify in 6lbr. 

to be clear my objective is:
I will have a QT UI application in RPi through which I would like to monitor and control the sensor nodes. From QT how can I communicate with 6lbr to receive data from the udp client running in CC2650

laurentderu, 2017-02-01:
@jvasanthan If your application runs on RPi, you don't need any fancy communication interface. Your application can open a UDP port and then communicates with the nodes through the BR (which as its name implies route packets :) )
Your application can extract the list of the nodes from the sensors page of 6LBR (it's valid xhtml), or the nodes can join your application using a well known IP address.

closed 2017-06-15 by laurentderu