cetic/6lbr #249: 6lbr.ip file missing

jbartlett12026, 2017-01-30:
Hello,
    I am using 6lbr with the following demo: http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
and I am not able to see the [bbbb::100] webpage. I looked at the 6lbr.ip file to see what was registered and the file is not there. Am I missing something here?

HW:  -Edge Router: Beaglebone Black with CC1310 LP as a slip radio --> flashed with slip-radio.bin
        -Mesh Nodes: CC1310 LP -->flashed with cc26xx-web-demo

Error file: Starting 6lbr
Log file: 
[image]

ronnie1233, 2017-02-01:
Did you set the RF Channel

jbartlett12026, 2017-02-01:
If this is done in the nvm.dat file then yes, I specified it to be 25, which is what I also specified on the CC1310.

jbartlett12026, 2017-02-01:
[image]

This is what the COM port for the CC1310 LP is saying. It looks like an error for the RF channel??

laurentderu, 2017-02-01:
@jbartlett12026 What is the content of your 6lbr.log file ? Looking at the output of the slip-radio, it seems the device is reseting constantly.

jbartlett12026, 2017-02-01:
I think you're correct. I have tried rebuilding and keep getting the following. It keeps restarting right before specifying the channel. Any thoughts? I built from slip-radio using **make TARGET=srf06-cc26xx BOARD=launchpad/cc1310** 

[image]

ronnie1233, 2017-02-02:
Yes you are on the right path here . It seems your slip radio does not  seem to  fetch BR-RDC  MAC address . Would recommend to look at  Project  config file .  If it is then it seems there is problem with ur submodule  in your RPI3 or BBB what ever you are using as border router .

jbartlett12026, 2017-02-02:
@ronnie1233 
 It was a problem with the USB-UART power I was using, which was causing it to loop and not start the radio. Now it appears to work before the putty crashes. Thank you for your response!

closed 2017-06-15 by laurentderu