cetic/6lbr #254: cc2538-cc2592 EM slip-radio doen't work

JungSuWhan, 2017-02-15:
I am using the 6lbr 1.4.x stable version as a slip radio with the cc2538-cc2592 em module. 
And I want to use the ti sensor tag as a 6lowpan mesh node.

Cc2538-cc2592 To make em a slip-radio, I compiled it in the 6lbr / example / ipv6 / slip-radio directory as follows.
#Setting# 
<plaform/cc2538dk/contiki-conf.h>
 #define UART_CONF_ENABLE            0
 #define SLIP_ARCH_CONF_USB         1
 #define DBG_CONF_USB                     1
 #define CC2538_RF_CONF_CHANNEL              25
 #define IEEE802154_CONF_PANID                    0xABCD

<ipv6/slip-radio/Makefile> 
SMALL 0
 
Build the Slip-Radio (examples/ipv6/slip-radio/) with “make TARGET=cc2538dk”
I succeeded in compiling. 

I installed the 6lbr 1.4.x on Beagle Bone Black and ran it using the cc2538-cc2592em module as a sleep radio.
It was normally connected to the 6lbr webpage.
Web page address bbbb :: 100

Cc2538-cc2592em module is normally mounted ttyACM0
However, the sensor tag does not normally find 6lbr. I downloaded the cc2531-slip-radio_contikimac.hex file to cc2531emk and tested it and it worked.
The pan id, channel, and prefix are set to 6lbr with the sensor tag.
Both the 6lbr and cc2538-cc2592em modules are set to have the same channel pan id prefix.
How can I compile the cc2538-cc2592emk as a sleep radio?

laurentderu, 2017-02-27:
Could you check the RDC compiled in your CC2538 slip-radio ? By default on Contiki it's ContikiMAC, however in the 6LBR tree, for historical reasons, it's NullRDC, and they are not compatible.
Also, check if the PAN ID configured is compatible.

JungSuWhan, 2017-02-28:
Thank you for your answer.
I've fixed the issue. Have a nice day

closed 2017-03-08 by laurentderu