cetic/6lbr #256: Example : UDP client

Maxilicious, 2017-02-24:
Hi,

I'd like to use the provided UDP client to connect to my 6LowPan gateway, but the page for demosntrating how to install the udp client is missing. Could you give me some directions on how to install this?

laurentderu, 2017-02-27:
Indeed, the UDP client demo page is still sadly empty... You can find the UDP demo client in examples/6lbr-demo/apps/udp-client and the corresponding server in examples/6lbr/apps/udp-server.

Note that the code is not the simple code possible, you should also have a look at the Contiki examples.

chenek, 2017-03-03:
Try to refer to http://sunmaysky.blogspot.tw/2017/03/how-to-run-contiki-udp-serverudp-client.html