cetic/6lbr #257: 6lbr on external network

MarCast, 2017-02-27:
Hello everyone, I have a working system with the 6lbr. Now I want to access 6lbr from an external network. Can someone help me? Thanks.

MarCast, 2017-02-27:
@laurentderu can you help us?

laurentderu, 2017-02-27:
I'm not sure what you mean as "external network", if you mean the LAN you should follow this page : https://github.com/cetic/6lbr/wiki/6LBR-Connection

If you mean a network outside your LAN, that would require either configuring your LAN router to advertise the route or using NAT.

MarCast, 2017-02-27:
Yes, I want to access 6lbr from another network. You can give me some help?

laurentderu, 2017-02-27:
If it's across the Internet, either you have IPv6 and you have received a /64 or /56 prefix and you can delegate a subnet to the WSN and connect directly from the outside world.

If not, you need to use NAT64 to make your network accessible using IPv4. You can use the integrated NAT64 module, https://github.com/cetic/6lbr/wiki/NAT64 Or you can use an external one, like SiXS or Freeness6.

MarCast, 2017-02-27:
Ok, I enabled NAT64. Are not yet able to access the 6lbr from an external network

laurentderu, 2017-02-27:
You need to configure it according to your network configuration, either by using DHCP or setting up a static address within your network range. But I'm not psychic, you should find your network configuration and do the setup yourself :)
If you have errors, you can report them here, but please provide enough information so one can investigate.

MarCast, 2017-02-27:
I am new to this field. I have enabled DHCP configuration of 6lbr. I from an address 192.169.233.241 still not accessible, while with the DHCP is not enabled me by an address 172.16.0.2

laurentderu, 2017-02-27:
could you check in /var/log/6lbr.log if the BR has started properly (and paste the log here), and also check in /var/log/6lbr.ip4 if he got an IPv4 address ?

MarCast, 2017-02-27:
6lbr.ip4 is 0.0.0.0 while lbr.log is working

MarCast, 2017-02-27:
This is my 6lbr.log:

********************************************************************************
lun 27 feb 2017, 16.02.25, CET : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyUSB0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www   -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins   
Contiki-develop-20161222 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2017-02-27 16:02:25.109064: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-develop-20161222)
2017-02-27 16:02:25.110923: INFO: ETH: 6LBR watchdog started (interval: 60)
2017-02-27 16:02:25.111808: INFO: NVM: NVM Magic : 2009
2017-02-27 16:02:25.111862: INFO: NVM: NVM Version : 5
2017-02-27 16:02:25.111881: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-02-27 16:02:25.111936: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-02-27 16:02:25.112028: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2017-02-27 16:02:25.112053: INFO: MAC: Using 'csma' mac driver
2017-02-27 16:02:25.116310: INFO: SLIP: SLIP started on /dev/ttyUSB0
2017-02-27 16:02:25.127860: INFO: SCMD: Started br-cmd process
2017-02-27 16:02:25.130123: INFO: BR-RDC: Reset SLIP Radio
2017-02-27 16:02:25.130304: INFO: BR-RDC: Fetching MAC address
2017-02-27 16:02:25.135158: ERROR: SLIP: Unknown command
2017-02-27 16:02:25.135532: INFO: BR-RDC: Got MAC: 00:12:4b:00:09:b7:9a:44
2017-02-27 16:02:26.132770: ERROR: SLIP: Unknown command
2017-02-27 16:02:26.132987: ERROR: BR-RDC: Set channel failed
2017-02-27 16:02:26.136690: ERROR: SLIP: Unknown command
2017-02-27 16:02:26.136905: ERROR: BR-RDC: Set PAN-ID failed
2017-02-27 16:02:26.137011: INFO: ETH: RAW Ethernet interface init
2017-02-27 16:02:26.196956: INFO: TAP: opened device /dev/eth0
2017-02-27 16:02:26.197206: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b8:27:eb:ce:b4:30  
          inet addr:192.168.233.241  Bcast:192.168.233.255  Mask:255.255.255.0
          inet6 addr: fe80::4d62:1c4c:b280:d739/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6146 errors:0 dropped:1 overruns:0 frame:0
          TX packets:559 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1888724 (1.8 MiB)  TX bytes:128874 (125.8 KiB)

2017-02-27 16:02:26.295185: INFO: TAP: Eth MAC address : b8:27:eb:ce:b4:30
2017-02-27 16:02:26.295455: INFO: ETH: Eth MAC address : b8:27:eb:ce:b4:30
2017-02-27 16:02:26.295620: INFO: LLSEC: Using 802.15.4 framer
2017-02-27 16:02:26.295735: INFO: LLSEC: Using 'nullsec' llsec driver
2017-02-27 16:02:26.296215: INFO: HTTP: Starting webserver on port 80
2017-02-27 16:02:26.297033: INFO: NODECFG: Node Config init
2017-02-27 16:02:26.297207: INFO: NODECFG: No node_config.conf file specified
2017-02-27 16:02:26.297335: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:9b7:9a44
2017-02-27 16:02:26.297683: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:9b7:9a44
2017-02-27 16:02:26.298023: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2017-02-27 16:02:26.298232: INFO: 6LBR: RA Daemon enabled
2017-02-27 16:02:26.298341: INFO: 6LBR: Checking addresses duplication
2017-02-27 16:02:26.336067: ERROR: PF: Trying to send empty packet
2017-02-27 16:02:38.301416: INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:9b7:9a44
2017-02-27 16:02:38.301865: INFO: 6LBR: Starting IP64
Starting DHCPv4
2017-02-27 16:02:38.302099: INFO: 6LBR: Starting as RPL ROUTER
2017-02-27 16:02:38.303553: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-02-27 16:02:38.303707: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-02-27 16:02:38.303836: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2017-02-27 16:02:38.304101: INFO: DNS: DNS proxy started
2017-02-27 16:02:38.304216: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-02-27 16:02:38.304384: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-02-27 16:02:38.304494: INFO: 6LBR: CETIC 6LBR Started
RD client started
2017-02-27 16:02:39.307585: INFO: 6LBR: Set IPv4 address : 192.168.233.241

MarCast, 2017-02-28:
@laurentderu

MarCast, 2017-02-28:
@banzo 
@jonas-simon

chenek, 2017-03-02:
Try to refer to http://sunmaysky.blogspot.tw/2017/02/how-to-do-settings-to-allow-access-6lbr.html

laurentderu, 2017-03-08:
The empty /var/log/6lbr.ip4 was a bug and is now solved in dev-1.4.x and develop branches.

closed 2017-03-08 by laurentderu