cetic/6lbr #260: Unable to build slip-radio on BOARD=launchpad/cc1310

electronut, 2017-03-06:
I am unable to build `slip-radio` on the CC1310 launchpad on a Raspberry Pi 3. The same builds fine under the `contiki` source tree. I have the latest version of `6lbr`.

```
~/6lbr/examples/ipv6/slip-radio $ make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 slip-radio.bin
  CC        ../../../cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/ccfg.c
../../../cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/ccfg.c:137:0: warning: "SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE" redefined [enabled by default]
 #define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0xC5       // Enable ROM boot loader
 ^
In file included from ../../../platform/srf06-cc26xx/./contiki-conf.h:419:0,
                 from <command-line>:0:
../../../platform/srf06-cc26xx/launchpad/cc1310/board.h:172:0: note: this is the location of the previous definition
 #define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0x00
 ^
../../../cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/ccfg.c:140:0: warning: "SET_CCFG_BL_CONFIG_BL_LEVEL" redefined [enabled by default]
 #define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x1        // Active high to open boot loader backdoor
 ^
In file included from ../../../platform/srf06-cc26xx/./contiki-conf.h:419:0,
                 from <command-line>:0:
../../../platform/srf06-cc26xx/launchpad/cc1310/board.h:173:0: note: this is the location of the previous definition
 #define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x01
 ^
  CC        ../../../cpu/cc26xx-cc13xx/dev/soc-rtc.c
../../../cpu/cc26xx-cc13xx/dev/soc-rtc.c: In function 'soc_rtc_init':
../../../cpu/cc26xx-cc13xx/dev/soc-rtc.c:103:25: error: 'INT_AON_RTC_COMB' undeclared (first use in this function)
   ti_lib_rom_int_enable(INT_AON_RTC_COMB);
                         ^
../../../cpu/cc26xx-cc13xx/dev/soc-rtc.c:103:25: note: each undeclared identifier is reported only once for each function it appears in
../../../Makefile.include:240: recipe for target 'obj_srf06-cc26xx/soc-rtc.o' failed
make: *** [obj_srf06-cc26xx/soc-rtc.o] Error 1
```

Under `contiki`:

```
~/contiki/examples/ipv6/slip-radio $ make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 slip-radio.bin
  CC        ../../../cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/ccfg.c
  CC        ../../../cpu/cc26xx-cc13xx/./ieee-addr.c
  AR        contiki-srf06-cc26xx.a
  CC        ../../../cpu/cc26xx-cc13xx/./fault-handlers.c
  CC        ../../../cpu/cc26xx-cc13xx/lib/cc13xxware/startup_files/startup_gcc.c
  CC        slip-radio.c
  LD        slip-radio.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff slip-radio.elf slip-radio.bin
rm slip-radio.co obj_srf06-cc26xx/fault-handlers.o obj_srf06-cc26xx/startup_gcc.o
```

I am not sure if it's advisable to use `slip-radio` built with `contiki` along with `6lbr` but in any case that is crashing `6lbr` for me.

chenek, 2017-03-07:
Yes, I always use slip-radio built with contiki.

laurentderu, 2017-03-08:
The master branch of 6LBR is based on Contiki 3.0 where there is no support of CC13xx. You should either switch to the develop branch, which is based on a quite recent snapshot of Contiki, or build your slip-radio from a Contiki repository.

electronut, 2017-03-09:
Thanks @laurentderu

closed 2017-03-09 by electronut