cetic/6lbr #262: Can't ping neighbour, route via (null)

tussock-cam, 2017-03-09:
I've got a 6lbr running on a beaglebone and a single device connected to it. I can see the status packet being sent from the node every minute when using a sniffer. The node is running web-demo with contiki @ a38cfc8 on a custom platform with a CC1310 radio. Initially it worked great, but after about 2 hours 6lbr can't talk to the node, even though it continually receives the node's packets.

I'm not 100% sure on my 6lbr version, unfortunately, but we installed with a cetic-6lbr_1.4.0_armhf.deb with MD5 284f6b385048acf4d7135107d0d95212  .

In the sensors list its status frequently changes from OK to NR, and its route is often removed. When it gets added, its state is typically:

> fd00::212:4b00:d2d:c202/128 via (null) 7678 s

The device's last seen never really goes above 60 seconds, so the BR is receiving its pings. However, if I try and ping the node, I get nothing back:

> root@beaglebone:~# ping6 -c 1 fd00::212:4b00:d2d:c202
> PING fd00::212:4b00:d2d:c202(fd00::212:4b00:d2d:c202) 56 data bytes
> 
> --- fd00::212:4b00:d2d:c202 ping statistics ---
> 1 packets transmitted, 0 received, 100% packet loss, time 0ms
> 

If I look at the sniffer, I don't see anything getting sent out. However, I do notice frequent pings from the node to 6lbr, which get replies. Not sure what these are, but am guessing it's part of normal RPL traffic?

Restarting the node doesn't fix it, and I haven't restarted 6lbr yet because I don't want to lose an opportunity to get some useful debugging done. Probably worth noting that web/coap also don't work when the ping doesn't. 

Any suggestions for narrowing down the cause? I've added my seemingly uninteresting 6lbr.log file below.

> Fri Mar 3 01:44:38 UTC 2017 : Starting 6LBR
> /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyS1 -R -t tap0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
> Contiki-6lbr-1.4.0 started with IPV6, RPL
> Rime started with address 1.2.3.4.5.6.7.8
> MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
> 2017-03-03 1:44:38.326142: NOTICE: 6LBR: Starting 6LBR version 1.4.0 (Contiki-6lbr-1.4.0)
> 2017-03-03 1:44:38.327675: INFO: ETH: 6LBR watchdog started (interval: 60)
> 2017-03-03 1:44:38.328569: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
> 2017-03-03 1:44:38.334083: INFO: PLUGIN: Initialising lwm2m-client
> 2017-03-03 1:44:38.334179: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
> 2017-03-03 1:44:38.335112: INFO: NVM: NVM Magic : 2009
> 2017-03-03 1:44:38.335167: INFO: NVM: NVM Version : 5
> 2017-03-03 1:44:38.335198: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2017-03-03 1:44:38.335482: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2017-03-03 1:44:38.335572: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
> 2017-03-03 1:44:38.335607: INFO: MAC: Using 'csma' mac driver
> 2017-03-03 1:44:38.349113: INFO: SLIP: SLIP started on /dev/ttyS1
> 2017-03-03 1:44:38.367943: INFO: SCMD: Started br-cmd process
> 2017-03-03 1:44:38.368236: INFO: BR-RDC: Reset SLIP Radio
> 2017-03-03 1:44:38.368350: INFO: BR-RDC: Fetching MAC address
> 2017-03-03 1:44:38.377846: ERROR: SLIP: Unknown command
> 2017-03-03 1:44:38.378372: INFO: BR-RDC: Got MAC: 00:12:4b:00:0a:27:c7:e4
> 2017-03-03 1:44:39.368321: INFO: ETH: TAP Ethernet interface init
> 2017-03-03 1:44:39.378127: INFO: TAP: opened device /dev/tap0
> 2017-03-03 1:44:39.379542: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
> 6lbr-ifup: Attach device tap0 to bridge br0
> br0       Link encap:Ethernet  HWaddr 90:59:af:5c:31:eb
>           inet addr:192.168.7.2  Bcast:192.168.7.255  Mask:255.255.255.0
>           inet6 addr: bbbb::9259:afff:fe5c:31eb/64 Scope:Global
>           inet6 addr: fe80::9259:afff:fe5c:31eb/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:7414 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:5002 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:1614953 (1.5 MiB)  TX bytes:475270 (464.1 KiB)
> 
> tap0      Link encap:Ethernet  HWaddr 02:0a:0b:0c:0d:0e
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
> 
> 2017-03-03 1:44:40.158303: INFO: ETH: Eth MAC address : 02:00:0a:27:c7:e4
> 2017-03-03 1:44:40.159993: INFO: LLSEC: Using 'nullsec' llsec driver
> 2017-03-03 1:44:40.161415: INFO: 6LBR: Security layer initialized
> 2017-03-03 1:44:40.162902: ERROR: SLIP: Unknown command
> 2017-03-03 1:44:40.164145: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:a27:c7e4
> 2017-03-03 1:44:40.166069: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:a27:c7e4
> 2017-03-03 1:44:40.167516: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
> 2017-03-03 1:44:40.169003: INFO: 6LBR: RA Daemon enabled
> 2017-03-03 1:44:40.169946: INFO: 6LBR: Checking addresses duplication
> 2017-03-03 1:44:50.165847: ERROR: PF: Trying to send empty packet
> 2017-03-03 1:44:52.183394: INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:a27:c7e4
> 2017-03-03 1:44:52.206063: INFO: 6LBR: Starting IP64
> Starting DHCPv4
> 2017-03-03 1:44:52.215726: INFO: 6LBR: Starting as RPL ROUTER
> 2017-03-03 1:44:52.226754: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2017-03-03 1:44:52.242688: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2017-03-03 1:44:52.256089: INFO: NODECFG: Node Config init
> 2017-03-03 1:44:52.257397: INFO: NODECFG: No node_config.conf file specified
> 2017-03-03 1:44:52.258502: INFO: HTTP: Starting webserver on port 80
> 2017-03-03 1:44:52.260609: INFO: UDPS: UDP server started
> CoAP push started
> CoAP server started
> 2017-03-03 1:44:52.281212: INFO: DNS: DNS proxy started
> 2017-03-03 1:44:52.292370: INFO: PLUGIN: Initialising lwm2m-client
> 2017-03-03 1:44:52.294360: INFO: LWM2M: LWM2M Client init
> 2017-03-03 1:44:52.295480: INFO: PLUGIN: Initialising dummy
> 2017-03-03 1:44:52.298707: INFO: DUMMY: Dummy init
> 2017-03-03 1:44:52.313474: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2017-03-03 1:44:52.324064: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2017-03-03 1:44:52.334161: INFO: 6LBR: CETIC 6LBR Started
> RD client started
> 2017-03-03 1:44:53.362388: INFO: 6LBR: Set IPv4 address : 192.168.7.14
> 2017-03-03 3:16:18.045804: WARN: HTTP: File '/favicon.ico' not found, from bbbb::8454:8ee0:57d:b7f8
> 2017-03-03 3:16:18.074573: WARN: HTTP: File '/favicon.ico' not found, from bbbb::8454:8ee0:57d:b7f8
> 2017-03-03 3:16:18.099612: WARN: WEB: Buffer overflow
> 2017-03-03 3:16:18.118029: WARN: WEB: Buffer overflow
> 
>

laurentderu, 2017-03-09:
The problem is that, for unknown reason, the neighbor entry for your node has been removed from the neighbor table of 6LBR, this is the (null) in the routing table. Because of that, the BR is no longer able to forward the packets towards your node as the next hop of the configured route does not exist anymore.

Now the root cause is not clear to me at all, and as you said, the 6lbr.log file does not show anything useful. To be able to analyse further, you should enable the traces in uip-ds6-nbr and maybe also in uip-ds6-route, rpl-dag and rpl-icmp6.
Note that 6LBR 1.4.x is still based on Contiki 3.0, on the other hand, 6LBR 1.5.x, the current develop branch, is based on the latest version of Contiki which contains updates and fixes related to the neighbor table management. It quite possible that you are observing a bug that is already solved, so before digging too much into the issue I would suggest to try with the latest version of the develop branch.

Btw, the ping you observe is in fact not related to RPL, it's a feature of the cc26xx-web-demo firmware which ping periodically its parent to get the current RSSI.

tussock-cam, 2017-03-09:
I'll test the develop branch and see if the issue remains. Just tried it now and the webserver fails after responding to one query, so need to figure out what's going on there. 

Will let you know how it goes.

tussock-cam, 2017-03-13:
So I've looked into this a bit today and saw it kind of happen again. This is against commit 744ad0cd1c0ee75136e09b11e8e7160208418683.

Soon after building the code, but before adding any debugging, I had the node connect, reliably ping 6lbr and be marked as NR. I couldn't tell why it had no route, but saw that it happened. It wasn't listed as `via (null)`, it just wasn't there. 

I then enabled printing in `uip-ds6-nbr.c`, `uip-ds6-route.c`, `rpl-icmp6.c` and `rpl.c` and tried to make it happen again. I couldn't get it to happen by itself, but if I removed the route via the web interface it'd remain as NR for about an hour, at which point I reset it. When the node rejoined it was given a route, so that obviously isn't the issue I noted before.

Anyway, looked through the code and saw that a route is only ever added when a node sends a DAO, which happens when it changes dag/parent. So if a route is removed for whatever reason, the node remains unreachable for quite a period of time, and as far as I can tell, it will have no idea. 

So... should a route be added when we receive data from a node? Or maybe a DAO requested? I'm not terribly familiar with RPL, so maybe this is a dumb suggestion, but it seems weird to be able to receive from a node for so long without giving it a route, and without it knowing it doesn't have a route.

I'm going to reset the test and leave it running overnight to see if I can get the other issue to happen again.

laurentderu, 2017-03-15:
In a RPL network, a downward route is only created when a valid DAO message is received. If a data packet is received from a node, a route is not created, otherwise you could have several downward routes towards the same node, some of them even not working as the radio conditions have changed.

In Contiki, the route created by RPL are protected, meaning that a route (and the corresponding next hop) will only be removed when an explicit No Path DAO (remove route message) or if the route lifetime has expired, or if there is a global or local repair.

In your case, the next hop has been deleted without removing the route, so it's most certainly a bug in the Contiki stack.

If you remove a route via the web interface, the end node does not know its route has been deleted so no new route will be created until the previous route lifetime is expired. To force recreation of the route you must trigger a global repair on the RPL page.

laurentderu, 2017-11-09:
This was solved in cfbc964d6d843774c71ee3539a5ce11a68350fa9

closed 2017-11-09 by laurentderu