cetic/6lbr #263: How

SHshenb, 2017-03-14:


SHshenb, 2017-03-14:
Where can i find the ipk code of 6lbr for openwrt?

zubair-apece, 2017-04-16:
Hi, did you find the solution? I need to install 6lbr in openwrt too

SHshenb, 2017-04-17:
I have installed 6lbr successfully ,but there still have some problem. How can i communicate with you to dicuss this problem and tell you how to isntall?

zubair-apece, 2017-04-17:
Thanks for your quick response and willingness to help. my email is kzubair.bd@gmail.com
I would really be grateful to you if you guide me to do so. I suppose we can help each other about this.

SHshenb, 2017-04-17:
I have sent a email for you.

knowic, 2017-04-17:
What is the problem you have meet? Also,I have got the Openwrt run with wrtnode board, but I have not get RA message correct from Openwrt
. My mail is knowic@gmail.com ,would you please send me mail so that we can have some discussion about the issue.

laurentderu, 2017-04-18:
I updated the OpenWRT documentation, see https://github.com/cetic/6lbr/wiki/OpenWRT-Installation

closed 2017-04-18 by laurentderu