cetic/6lbr #264: Gateway Contiki OS

MarCast, 2017-03-14:
Hi everyone, I would like to implement a firmware of gateway. I am using a raspberry and two sensors, one for the slip-radio and another for the CC26xx-web-demo. Should I bring the Raspberry on the external network and the slip-radio must become a gateway. Could someone please tell me where I can find the file gateway of 6lbr or some help?

MarCast, 2017-03-15:
@laurentderu

laurentderu, 2017-03-15:
If you are looking for a package to install, unfortunately there are none available currently until we define the distribution we will officially test and support. In the mean tile you have to manually compile and install 6LBR, see https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration

If you're looking for the 6LBR code to modify it, it's available here : https://github.com/cetic/6lbr/tree/develop/examples/6lbr

chenek, 2017-03-22:
Do you mean to run 6lbr on CC26xx?

closed 2017-03-23 by laurentderu