cetic/6lbr #268: problem run contiki examples in nodes

fzy-Line, 2017-03-20:
Hi,
I run 6lbr in my Raspberry Pi and connect it with a CC2538dk which use as the Slip-Radio. If I run the 6lbr-demo in nodes, I can see the network runs well. But if I run a contiki example in a node , there are no data about Parent, Up PRR and Down PRR shows in the webserver sensor page.

Who can help me solve this problem?

fzy-Line, 2017-03-20:
just like this:
[image]

stnitomy, 2017-03-31:
Are you getting any error in PRR stating Data column(s) for axis #0 cannot be of type string???

laurentderu, 2017-03-31:
That errors means that no data is available to populate the graphes. Without the cetic-6lbr-client on the node, the BR can only retrieve the hop count information and the bandwidth per node. PRR, Parent Switch and node tree will be empty.

fzy-Line, 2017-04-05:
@stnitomy : Hello, I'm sorry, I don't get the error.  I don't use the Cooja simulation and I run the 6lbr in my Raspberry Pi.

fzy-Line, 2017-04-05:
@laurentderu : Thanks for you response. You mean that I should run the cetic-6lbr-client on the node, but I don‘t know how to run the cetic-6lbr-client. Because I use the 6lbr-demo which include many app, such as coap, udp-client and webserver, but I could't find the code that run as cetic-6lbr-client. Can you give me some help? Thanks again!

msolters, 2017-04-06:
Have you looked at something like https://github.com/contiki-os/contiki/blob/master/examples/cc26xx/cc26xx-web-demo/cetic-6lbr-client.c

fzy-Line, 2017-04-06:
@msolters :+1: Thank you for the perfect answer to my question. It seems that I don't add the cetic-6lbr-client to my node app, so it don't send any information about the node to the 6lbr service which make that shows blank in the webserver sensor page. Thank you again!

closed 2017-04-06 by fzy-Line

stnitomy, 2017-04-06:
@fzy-Line; If you are aware of the steps to run cetic-6lbr-client, could you kindly share it. Coy when I tried running simulation I was getting error.