cetic/6lbr #271: recipe for target 'obj_cooja/cooja_mtarch.o' failed

stnitomy, 2017-03-21:
I was going through the installation of 6lbr integerated WSN from 
https://github.com/cetic/6lbr/wiki/Other-Linux-Software-Configuration

My current JAVA version is:
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)

Also i have edited /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait dwc_otg.speed=1

Have disabled getty on Raspberry Pi serial line
sudo systemctl stop serial-getty@ttyAMA0.service 

When i am trying to run make cooja-small, I am encountering this error:

BUILD SUCCESSFUL
Total time: 2 seconds
~/6lbr/examples/6lbr/test
if [ ! -f ~/.cooja.user.properties ]; then cp ./conf-templates/cooja.user.properties.example ~/.cooja.user.properties; fi
if [ `whoami` = root ] && [ ! -f /root/.cooja.user.properties ] ; then cp ./conf-templates/cooja.user.properties.example /root/.cooja.user.properties; fi
if ! grep -Fq "serial2pty" ~/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/serial2pty/" ~/.cooja.user.properties; fi
if ! grep -Fq "radiologger-headless" ~/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/radiologger-headless/" ~/.cooja.user.properties; fi
if [ `whoami` = root ] && ! grep -Fq serial2pty /root/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/serial2pty/" /root/.cooja.user.properties; fi
if [ `whoami` = root ] && ! grep -Fq radiologger-headless /root/.cooja.user.properties ; then sed -i "/^DEFAULT_PROJECTDIRS=/ s/\$/\;\[APPS_DIR\]\/radiologger-headless/" /root/.cooja.user.properties; fi
java -jar ../../../tools/cooja/dist/cooja.jar -quickstart=`cat ./coojagen/output/LASTFILE`
 INFO [main] (Cooja.java:1315) - > Starting Cooja
 INFO [main] (Cooja.java:2856) - External tools default settings: /external_tools_linux.config
 INFO [main] (Cooja.java:2886) - External tools user settings: /home/pi/.cooja.user.properties
 INFO [main] (Simulation.java:431) - Simulation random seed: 9194033042561454866
 INFO [main] (CompileContiki.java:140) - > make TARGET=cooja clean 
 INFO [main] (CompileContiki.java:140) - > make slip-radio.cooja TARGET=cooja 
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:171: Error: bad instruction `popl %esi'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:172: Error: bad instruction `popl %edx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:173: Error: bad instruction `popl %ecx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:174: Error: bad instruction `popl %ebx'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:175: Error: bad instruction `popl %eax'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:176: Error: bad instruction `leave'
FATAL [main] (ContikiMoteType.java:350) - >> /tmp/cc71uJhT.s:177: Error: bad instruction `ret'
FATAL [main] (ContikiMoteType.java:350) - >> make: *** [obj_cooja/cooja_mtarch.o] Error 1
FATAL [main] (ContikiMoteType.java:350) - >> ../../../Makefile.include:240: recipe for target 'obj_cooja/cooja_mtarch.o' failed
FATAL [main] (ContikiMoteType.java:350) - >> Process returned error code 2
FATAL [main] (ContikiMoteType.java:353) - Compilation error: Bad return value

What might be the reason for this error? Did I miss something in configuration

stnitomy, 2017-03-21:
Cooja Error code:

INFO [AWT-EventQueue-0] (MessageListUI.java:257) - slip-net.c:50:12: warning: unused variable 'i' [-Wunused-variable]
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -    uint16_t i;
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -             ^
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        no-framer.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        slip-radio-cooja.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        ../../../apps/slip-cmd/cmd.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        ../../../apps/slip-cmd/packetutils.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        ../../../platform/cooja/lib/simEnvChange.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        ../../../platform/cooja/sys/cooja_mt.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) -   CC        ../../../platform/cooja/sys/cooja_mtarch.c
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s: Assembler messages:
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:142: Error: bad instruction `pushl %eax'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:143: Error: bad instruction `pushl %ebx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:144: Error: bad instruction `pushl %ecx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:145: Error: bad instruction `pushl %edx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:146: Error: bad instruction `pushl %esi'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:147: Error: bad instruction `pushl %edi'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:148: Error: bad instruction `pushl %ebp'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:149: Error: bad instruction `pushl %ebp'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:156: Error: bad instruction `movl [r3],%eax'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:161: Error: bad instruction `movl (%eax),%ebx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:162: Error: bad instruction `movl %esp,(%eax)'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:163: Error: bad instruction `movl %ebx,%esp'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:168: Error: bad instruction `popl %ebp'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:169: Error: bad instruction `popl %ebp'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:170: Error: bad instruction `popl %edi'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:171: Error: bad instruction `popl %esi'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:172: Error: bad instruction `popl %edx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:173: Error: bad instruction `popl %ecx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:174: Error: bad instruction `popl %ebx'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:175: Error: bad instruction `popl %eax'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:176: Error: bad instruction `leave'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - /tmp/cc6TEMgI.s:177: Error: bad instruction `ret'
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - ../../../Makefile.include:240: recipe for target 'obj_cooja/cooja_mtarch.o' failed
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - make: *** [obj_cooja/cooja_mtarch.o] Error 1
 INFO [AWT-EventQueue-0] (MessageListUI.java:257) - Process returned error code 2

laurentderu, 2017-03-23:
Are you trying to run Cooja on a RPi ? Cooja mote are only supported on x86 architecture, not ARM, unless you implement the low level code.

stnitomy, 2017-03-31:
@laurentderu : THanks for the reply. I switched to linux host running 6lbr and trying to simulate WSN using cooja. I have got a confusion in that. For some tabs under sensor feature list, I am getting an error stating Data column(s) for axis #0 cannot be of type string. How can I trigger the DAG in this Cooja setup??

laurentderu, 2017-03-31:
See https://github.com/cetic/6lbr/issues/268

closed 2017-03-31 by laurentderu