cetic/6lbr #272: Problem to build 6lbr for cc26xx using latest 6lbr code

chenek, 2017-03-22:
I have see the following problem to build 6lbr for cc26xx using latest 6lbr code. It seems that cc26xxware are out of sync.

user@instant-contiki:~/6lbr_new/examples/6lbr$ make TARGET=srf06-cc26xx BOARD=srf06/cc26xx all
rm -rf bin_srf06-cc26xx
mkdir -p bin_srf06-cc26xx
make clean
make[1]: Entering directory `/home/user/6lbr_new/examples/6lbr'
rm -f *~ *core core *.srec \
	*.lst *.map \
	*.cprg *.bin *.data contiki*.a *.firmware core-labels.S *.ihex *.ini \
	*.ce *.co
rm -rf *.srf06-cc26xx symbols.c symbols.h *.d *.elf *.hex
rm -rf obj_srf06-cc26xx
make[1]: Leaving directory `/home/user/6lbr_new/examples/6lbr'
make bin_srf06-cc26xx/cetic_6lbr_smart_bridge.bin
make[1]: Entering directory `/home/user/6lbr_new/examples/6lbr'
make CETIC_6LBR_SMARTBRIDGE=1 WITH_UDPCLIENT=0
make[2]: Entering directory `/home/user/6lbr_new/examples/6lbr'
mkdir obj_srf06-cc26xx
  CC        ./../../cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c
./../../cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c:137:0: warning: "SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE" redefined
 #define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0xC5       // Enable ROM boot loader
 ^
In file included from ./../../platform/srf06-cc26xx/./contiki-conf.h:419:0,
                 from <command-line>:0:
./../../platform/srf06-cc26xx/srf06/cc26xx/board.h:245:0: note: this is the location of the previous definition
 #define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0x00
 ^
./../../cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c:140:0: warning: "SET_CCFG_BL_CONFIG_BL_LEVEL" redefined
 #define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x1        // Active high to open boot loader backdoor
 ^
In file included from ./../../platform/srf06-cc26xx/./contiki-conf.h:419:0,
                 from <command-line>:0:
./../../platform/srf06-cc26xx/srf06/cc26xx/board.h:246:0: note: this is the location of the previous definition
 #define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x01
 ^
  CC        ./apps/6lbr-webserver-lite/httpd-simple.c
  CC        ./apps/6lbr-webserver-lite/webserver-lite.c
  CC        ./../../platform/srf06-cc26xx/./contiki-main.c
  CC        ./../../core/lib/sensors.c
  CC        ./../../core/dev/leds.c
  CC        ./../../platform/srf06-cc26xx/srf06/leds-arch.c
./../../platform/srf06-cc26xx/srf06/leds-arch.c: In function 'leds_arch_init':
./../../platform/srf06-cc26xx/srf06/leds-arch.c:59:3: warning: implicit declaration of function 'GPIO_clearMultiDio' [-Wimplicit-function-declaration]
   ti_lib_gpio_clear_multi_dio(BOARD_LED_ALL);
   ^
./../../platform/srf06-cc26xx/srf06/leds-arch.c: In function 'leds_arch_set':
./../../platform/srf06-cc26xx/srf06/leds-arch.c:77:5: warning: implicit declaration of function 'GPIO_setDio' [-Wimplicit-function-declaration]
     ti_lib_gpio_set_dio(BOARD_IOID_LED_1);
     ^
  CC        ./../../platform/srf06-cc26xx/srf06/srf06-sensors.c
  CC        ./../../platform/srf06-cc26xx/srf06/button-sensor.c
./../../platform/srf06-cc26xx/srf06/button-sensor.c: In function 'button_press_handler':
./../../platform/srf06-cc26xx/srf06/button-sensor.c:90:5: warning: implicit declaration of function 'GPIO_readDio' [-Wimplicit-function-declaration]
     if(ti_lib_gpio_read_dio(BOARD_IOID_KEY_SELECT) == 0) {
     ^
./../../platform/srf06-cc26xx/srf06/button-sensor.c: In function 'config_buttons':
./../../platform/srf06-cc26xx/srf06/button-sensor.c:196:5: warning: implicit declaration of function 'GPIO_clearEventDio' [-Wimplicit-function-declaration]
     ti_lib_gpio_clear_event_dio(key);
     ^
  CC        ./../../platform/srf06-cc26xx/srf06/board.c
./../../platform/srf06-cc26xx/srf06/board.c: In function 'lpm_handler':
./../../platform/srf06-cc26xx/srf06/board.c:53:3: warning: implicit declaration of function 'GPIO_clearDio' [-Wimplicit-function-declaration]
   ti_lib_gpio_clear_dio(BOARD_IOID_ALS_PWR);
   ^
  CC        ./../../platform/srf06-cc26xx/srf06/als-sensor.c
./../../platform/srf06-cc26xx/srf06/als-sensor.c: In function 'config':
./../../platform/srf06-cc26xx/srf06/als-sensor.c:68:7: warning: implicit declaration of function 'GPIO_setDio' [-Wimplicit-function-declaration]
       ti_lib_gpio_set_dio(BOARD_IOID_ALS_PWR);
       ^
./../../platform/srf06-cc26xx/srf06/als-sensor.c:73:7: warning: implicit declaration of function 'GPIO_clearDio' [-Wimplicit-function-declaration]
       ti_lib_gpio_clear_dio(BOARD_IOID_ALS_PWR);
       ^
  CC        ./../../cpu/cc26xx-cc13xx/./clock.c
  CC        ./../../cpu/cc26xx-cc13xx/./rtimer-arch.c
  CC        ./../../cpu/cc26xx-cc13xx/dev/soc-rtc.c
./../../cpu/cc26xx-cc13xx/dev/soc-rtc.c: In function 'soc_rtc_init':
./../../cpu/cc26xx-cc13xx/dev/soc-rtc.c:103:25: error: 'INT_AON_RTC_COMB' undeclared (first use in this function)
   ti_lib_rom_int_enable(INT_AON_RTC_COMB);
                         ^
./../../cpu/cc26xx-cc13xx/dev/soc-rtc.c:103:25: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [obj_srf06-cc26xx/soc-rtc.o] Error 1
make[2]: Leaving directory `/home/user/6lbr_new/examples/6lbr'
make[1]: *** [bin_srf06-cc26xx/cetic_6lbr_smart_bridge.bin] Error 2
make[1]: Leaving directory `/home/user/6lbr_new/examples/6lbr'
make: *** [all] Error 2

laurentderu, 2017-03-23:
Tanks for reporting, it seems that during the latest merge the update of the CC13xx and CC26xx libraries were dropped.
I fixed the versions included and updated 6LBR code for the new TI API, could you give it a try ?

chenek, 2017-03-24:
Just have a try to build latest code and it works fine. Thanks for fixing this issue!

closed 2017-03-24 by chenek