cetic/6lbr #273: Cetic_6lbr_router

MarCast, 2017-03-23:
Hi everyone, I have some problem. I working with CC1310F128 and SmartRF06EB with Instant Contiki. I would like to connect to 6lbr with ENC28J60. I tested a cetic_6lbr_router but on monitor serial, I don't see nothing and don't connect with [bbbb::100]. Can you help me?

laurentderu, 2017-03-23:
It should work on the latest develop, but you need to configure the NVM configuration according to your board and also maybe the ENC28J60 pinout.
If you don't have output on the serial at all after resetting the board, it either means your firmware is not build for the right hardware or the NVM configuration is wrong. See https://github.com/cetic/6lbr/issues/244 changing the NVM from 'int' to 'ext' solved the problem.

MarCast, 2017-03-23:
The setting of the board.h is done well, I downloaded the 6lbr and I generated the file.bin. I do another or enough?

MarCast, 2017-03-23:
I from error in construction in the file.bin

chenek, 2017-03-24:
The latest commit fixes build issue. You can checkout latest code to build again.

MarCast, 2017-03-24:
How?

MarCast, 2017-03-27:
@laurentderu @chenek

laurentderu, 2017-03-27:
You can change the NVM setting using the following option in your Makefile :

    CC26XX_NVM=int

By default it is set to 'int', but you should switch to 'ext' and see if it goes better

The ENC28J60 pinout can be modified using the following macros :

    CC26XX_ENC28J60_CONF_CLK_PORT
    CC26XX_ENC28J60_CONF_MOSI_PORT
    CC26XX_ENC28J60_CONF_MISO_PORT
    CC26XX_ENC28J60_CONF_CS_PORT

The default values are respectively :

    BOARD_IOID_SPI_SCK
    BOARD_IOID_SPI_MOSI
    BOARD_IOID_SPI_MISO
    CC26XX_ENC28J60_CONF_CS_PORT

As @chenek  said, there were compilation errors, you should refresh your local copy of the 6LBR repo to have the fix.

MarCast, 2017-03-27:
@laurentderu What is the Makefile? I don't find

chenek, 2017-03-27:
Isn't it at https://github.com/cetic/6lbr/blob/develop/examples/6lbr-demo/Makefile

MarCast, 2017-03-27:
But the file cetic_6lbr_router must be generated under: 6LBR/examples/6lbr?

chenek, 2017-03-27:
Yes

MarCast, 2017-03-27:
I don't found the Makefile that I should change

chenek, 2017-03-28:
Anyway, if you don't know how to define those SPI pins in Makefile, you can revise them in enc28j60-arch-gpio.c under \6lbr\examples\6lbr\platform\srf06-cc26xx

MarCast, 2017-03-28:
It works!! Thanks!

closed 2017-03-28 by MarCast