cetic/6lbr #276: HTTP server not responding after first request.

omerfirmak, 2017-03-26:
I got 6lbr set up on my Raspberry Pi using a Sensortag CC2650 kit as slip-radio.

I can access the web interface through [bbbb::100] when I try the first time. But if I click any tabs or try to reload the page Firefox cannot establish a connection to the server.

I can ping the [bbbb::100] address with no problem at any stage. 6lbr.log at LOG_LEVEL=3 or 6lbr.err shows to sign of a problem.

2017-03-26 14:23:26.713938: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-base-develop-20170120-2305-g98a3eac)

laurentderu, 2017-03-27:
This looks like a bridging issue. Are you using Bridging or Tap mode. If it's bridging, could you check if you have correctly enabled the multicast snooping workaround and if it is enabled ?
The workaround is either enabled in /etc/network/interfaces with the line 

    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping

or in 6lbr.conf file using 

    BRIDGE_DISABLE_SNOOPING=1

Which version of the RPi are you using as it might be possible that the Ethernet Controller can not support Promiscuous mode

omerfirmak, 2017-04-13:
I am using the default configuration. 

MODE=ROUTER
RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3 #INFO and above only

And its a Raspberry Pi 2.

Setting BRIDGE_DISABLE_SNOOPING=1 didnt help

omerfirmak, 2017-04-13:
Event with RAW_ETH=1 BRIDGE=0 same thing happens.

laurentderu, 2017-04-18:
That's a strange behavior, with Raw Ethernet you should not have any connectivity issues like that. Could you provide a pcap file when you ping the BR and when you try to access (unsuccessfully) the BR ?

marhaus, 2017-04-20:
I have the exact same behaviour.

I have attached:
- `6lbr.conf` (.txt)
- `http-death.pcapng` (.zip)
- `6lbr.log` (.txt)

My 6lbr node is `bbbb::200` and my client (PC) is `bbbb::201`. 6lbr node CC2650 MAC address ends with "0a:4a:3a:06".

What I did:

1. `ping bbbb::200` (three pings) @ 14.45.07
2. open `http://[bbbb::200]` @ 14.45.27
3. open `http://[bbbb::200]` @ 14.45.39

1 worked fine, 2 too, 3 failed.

New ping afterwards (not in logs) works fine. All following attempts to open the web page fail though.

I temporarily increased the 6lbr log level to 3 although this is not visible in the attached conf file. There are a couple of other devices on the RF network so there are spurious log entries in `6lbr.log` that are not relevant here.

/Tomas G.

[6lbr.conf.txt](https://github.com/cetic/6lbr/files/943124/6lbr.conf.txt)
[6lbr.log.txt](https://github.com/cetic/6lbr/files/943125/6lbr.log.txt)
[http-death.zip](https://github.com/cetic/6lbr/files/943126/http-death.zip)

omerfirmak, 2017-04-26:
Here is my pcap file. It shows same behaviour with the one provided above.

[6lbr_http.zip](https://github.com/cetic/6lbr/files/957647/6lbr_http.zip)

Hope to hear back from you soon.

lcharpenwi6labs, 2017-05-03:
Hello
I am currently testing the new version of the 6lbr and I have exactly the same issue.
I can only connect to one page of the web server (any page seems to work). Any other action on the webpage fails then.

I can see this in the log:
2017-05-03 8:44:00.967872: DEBUG: HTTP: Request for '/index.html' from bbbb::6565:4583:e79d:bc38
2017-05-03 8:44:01.054357: DEBUG: HTTP: Request for '/6lbr_layout.css' from bbbb::6565:4583:e79d:bc38
2017-05-03 8:44:01.055857: DEBUG: WEB: Sending file /usr/lib/6lbr/www/6lbr_layout.css

And then, the web connection is dead...
I tested also with several browsers just in case but still the same result.

Thanks for your help

marco-pistone, 2017-05-22:
Hello,
i solved on my beglebone only modifying the outputbuf size in examples/6lbr/apps/6lbr-webserver/httpd.h

struct httpd_state {
  struct timer timer;
  struct psock sin, sout;
  struct pt outputpt;
  char inputbuf[HTTPD_PATHLEN + 24];
  char *query;
#if CONTIKI_TARGET_NATIVE
		//char outputbuf[UIP_TCP_MSS];
		char outputbuf[512];
  int fd;
  int len;
  int to_send;
#endif
  char filename[HTTPD_PATHLEN];
  char request_type;
  struct httpd_cgi_call *script;
  char state;
};

Hope this clue help to fix this bug.

P.S.
i need this fix only for the beaglebone black code, instead using the 6lbr on a debian virtual machine i never notice this issue.

laurentderu, 2017-05-24:
This problem is a crazy one, the problem never occurs on Linux, OpenWRT, MacOS, ... For an unknown reason, the TCP connection for the web server ends up in a wrong state and every new request gets back a RST. Also the problem occurs only if the output data is greater than UIP_TCP_MSS - 6, below the web server works fine.

A wild guess is that there is a buffer overflow somewhere in the IP stack when RPL extension header is added temporarily.

I will change the buffer size to UIP_TCP_MSS - 8 and hope that the problem is solved due to that.

marhaus, 2017-05-24:
Not completely true, my system is running Linux (Debian 7, kernel 3.7). It's on `armv5tejl` if that helps?

But thanks very much for your help!

laurentderu, 2017-05-24:
Actually I meant Linux on x86 :)

Both dev-1.4.x and develop branches have been updated with the patch.

Thanks for the investigation @marco-pistone, I never thought the problem was in that part of the code.

marhaus, 2017-05-24:
Fantastic, thanks! Both of you :)

lcharpenwi6labs, 2017-05-24:
Thank you very much for this tricky one!

closed 2017-06-15 by laurentderu