cetic/6lbr #277: cetic_6lbr_router help

MarCast, 2017-03-27:
Hi everyone, I generate cetic_6lbr_router on my Raspberry but don't works. I have a CC1310F128 and ENC28J60. Can you help me? I don't see nothing on UART. The my board.h is correctly.

laurentderu, 2017-03-27:
Please do not open many issues, it will not give you faster answers, on the contrary, as the history of the issue is lost.

closed 2017-03-27 by laurentderu

MarCast, 2017-03-27:
@laurentderu  One question, but with CC1310F128 and ENC28J60, Can work together? The cetic_6lbr_router.bin can work with CC1310F128?

laurentderu, 2017-03-27:
Yes, I have tested the CC1310 on a SmartRF-06 board some time ago, it worked without troubles.

MarCast, 2017-03-27:
5X5? @laurentderu

laurentderu, 2017-03-27:
The CC1310EMK is a 7x7 I believe. The support for 5x5 is ongoing in Contiki and I haven't tested it. There are two ongoing pull request for other CC13xx hardware on Contiki but I haven't followed them closely.

MarCast, 2017-03-27:
I have a board 5x5? How can I do?

MarCast, 2017-03-27:
I do not think that I can not use it for the CC1310 5x5..Don't you think @laurentderu ?