cetic/6lbr #279: COAP DTLS example client

Gr3yh0und, 2017-03-28:
Hi,

I've successfully compiled the COAP example for my CC2538DK and flashed it. When I use the cf-client found [here](https://repo.eclipse.org/content/repositories/californium-releases/org/eclipse/californium/cf-client/1.0.1/) with `java -jar cf-client-1.0.1.jar DISCOVER coap://fd00::212:4b00:615:a86b/` it works perfectly fine:

> ==[ CoAP Response ]============================================
> MID    : 63376
> Token  : 6f
> Type   : ACK
> Status : 2.05
> Options: {"Content-Format":"application/link-format", "Block2":"(szx=2/64, m=true, num=0)"}
>Payload: 281 Bytes
> ...

Now when I enable WITH_TINYDTLS and WITH_DTLS_COAP the board starts up mentioning:
`make TARGET=cc2538dk WITH_COAPSERVER=1 WITH_TINYDTLS=1 WITH_DTLS_COAP=1`

> Started DTLS retransmit process
> COAP-DTLS listening on port 5683
> CoAP push started
> CoAP server started
> RD client started
> ...
> Response from the server: '254'

And when I use `java -jar cf-client-1.0.1.jar DISCOVER coaps://fd00::212:4b00:615:a86b/` the client is continuing to make a handshake. Obviously unsuccessfully... When I check  `6lbr/apps/er-coap/er-coap-dtls.c` I find the identity `Client_identity` and PSK `secretPSK`. However if I use the cf-client with addition of `-psk` and enter it manually, nothing changes. Where do I need to change the PSK/identity on the client side? 

When I try to use `coap` instead of `coaps` the server responds on UART with:

> dtls_handle_message: PEER NOT FOUND
> peer addr: :38845

It is also quite hard to find a COAP Client supporting DTLS... Could you please help/update the tutorial with some links to .jar's or whatever one needs to make a client request? When I build californium from source the `cf-client` isn't built anymore. And i used the `certs` folder from that.

Thanks!

Cheers

Gr3yh0und, 2017-03-28:
I did came a bit closer. After checking out the [tools-repository](https://github.com/eclipse/californium.tools), building the `cf-client` with `mvn clean install` and copying it to the `/run` directory to get the `certs` folder path right I am now on the newest client. 

I also found out that the CoAP DTLS server on the node is running on the port 5683 while the default port on the client is 5684. That's the cause why no connection was possible...

So I am now running `java -jar cf-client-1.1.0-SNAPSHOT.jar GET coaps://[fd00::212:4b00:615:a86b]:5683/.well-known/core` and see the response on the UART:

> dtls_handle_message: PEER NOT FOUND
> peer addr: :44182
> got packet 22 (103 bytes)
> DEBG receive header: (u bytes):
> 00000000 16 FE FD 00 00 00 00 00  00 00 00 00 5A
> DEBG receive unencrypted: (u bytes):
> 00000000 01 00 00 4E 00 00 00 00  00 00 00 4E FE FD 58 DA
> 00000010 5E 98 BF D4 BF 9E D6 61  CF 12 23 52 4D CD 5E 13
> 00000020 61 E4 F7 FB 05 A2 80 71  2C E6 E2 DA 5C 18 00 00
> 00000030 00 04 C0 AE C0 23 01 00  00 20 00 0A 00 08 00 06
> 00000040 00 17 00 18 00 19 00 0B  00 02 01 00 00 13 00 03
> 00000050 02 02 00 00 14 00 03 02  00 02
> received handshake packet of type: client_hello (1)
> handle handshake packet of type: client_hello (1)
> DEBG create cookie: (u bytes): 447D9B7B54F86D6CF463BDC31A946FF8
> DEBG compare with cookie: (u bytes):
> cookie len is 0!
> send handshake packet of type: hello_verify_request (3)
> DEBG send header: (u bytes):
> 00000000 16 FE FD 00 00 00 00 00  00 00 00 00 1F
> DEBG send unencrypted: (u bytes):
> 00000000 03 00 00 13 00 00 00 00  00 00 00 13
> DEBG send unencrypted: (u bytes):
> 00000000 FE FD 10 44 7D 9B 7B 54  F8 6D 6C F4 63 BD C3 1A
> 00000010 94 6F F8
> server hello verify was sent

while the bash gives me something a bit bulky:

>  1 CONFIG [InMemoryConnectionStore]: Created new InMemoryConnectionStore [capacity: 150,000, connection expiration threshold: 1,800s] - (org.eclipse.californium.scandium.dtls.InMemoryConnectionStore.java:194) <init>() in thread main at (2017-03-28 15:01:12)
>  1 CONFIG [DTLSConnector]: Cannot determine MTU of network interface, using minimum MTU [1280] of IPv6 instead - (org.eclipse.californium.scandium.DTLSConnector.java:277) start() in thread main at (2017-03-28 15:01:12)
> 11 CONFIG [DTLSConnector$Worker]: Starting worker thread [DTLS-Receiver-0.0.0.0/0.0.0.0:44182] - (org.eclipse.californium.scandium.DTLSConnector$Worker.java:-1) run() in thread DTLS-Receiver-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
>  1 INFO [DTLSConnector]: DTLS connector listening on [0.0.0.0/0.0.0.0:44182] with MTU [1,280] using (inbound) datagram buffer size [16,474 bytes] - (org.eclipse.californium.scandium.DTLSConnector.java:235) start() in thread main at (2017-03-28 15:01:12)
> 10 CONFIG [DTLSConnector$Worker]: Starting worker thread [DTLS-Sender-0.0.0.0/0.0.0.0:44182] - (org.eclipse.californium.scandium.DTLSConnector$Worker.java:-1) run() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 10 FINER [DTLSConnector]: Sending application layer message to peer [/fd00:0:0:0:212:4b00:615:a86b:5683] - (org.eclipse.californium.scandium.DTLSConnector.java:1130) sendMessage() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 10 FINER [DTLSSession]: Setting MTU for peer [/fd00:0:0:0:212:4b00:615:a86b:5683] to 1,280 bytes - (org.eclipse.californium.scandium.dtls.DTLSSession.java:214) setMaxTransmissionUnit() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 10 FINER [DTLSSession]: Setting maximum fragment length for peer [/fd00:0:0:0:212:4b00:615:a86b:5683] to 1,227 bytes - (org.eclipse.californium.scandium.dtls.DTLSSession.java:598) determineMaxFragmentLength() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 10 FINE [Connection]: Handshake with [/fd00:0:0:0:212:4b00:615:a86b:5683] has been started - (org.eclipse.californium.scandium.dtls.Connection.java:865) handshakeStarted() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 10 FINER [DTLSConnector]: Sending flight of 1 message(s) to peer [/fd00:0:0:0:212:4b00:615:a86b:5683] using 1 datagram(s) of max. 1,280 bytes - (org.eclipse.californium.scandium.DTLSConnector.java:1257) sendFlight() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:12)
> 13 FINE [DTLSConnector]: Re-transmitting flight for [/fd00:0:0:0:212:4b00:615:a86b:5683], [3] retransmissions left - (org.eclipse.californium.scandium.DTLSConnector.java:112) handleTimeout() in thread DTLS RetransmitTask-1 at (2017-03-28 15:01:13)
> 13 FINER [DTLSConnector]: Sending flight of 1 message(s) to peer [/fd00:0:0:0:212:4b00:615:a86b:5683] using 1 datagram(s) of max. 1,280 bytes - (org.eclipse.californium.scandium.DTLSConnector.java:1350) sendFlight() in thread DTLS RetransmitTask-1 at (2017-03-28 15:01:13)
> 10 FINER [DTLSConnector]: Sending application layer message to peer [/fd00:0:0:0:212:4b00:615:a86b:5683] - (org.eclipse.californium.scandium.DTLSConnector.java:1130) sendMessage() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:14)
> 10 FINER [DTLSSession]: Setting MTU for peer [/fd00:0:0:0:212:4b00:615:a86b:5683] to 1,280 bytes - (org.eclipse.californium.scandium.dtls.DTLSSession.java:214) setMaxTransmissionUnit() in thread DTLS-Sender-0.0.0.0/0.0.0.0:44182 at (2017-03-28 15:01:14)
[... cut ...]
> FINE [Connection]: Handshake with [/fd00:0:0:0:212:4b00:615:a86b:5683] has been started
> FINER [DTLSConnector]: Sending flight of 1 message(s) to peer [/fd00:0:0:0:212:4b00:615:a86b:5683]
> FINE [DTLSConnector]: Re-transmitting flight for [/fd00:0:0:0:212:4b00:615:a86b:5683], [3] retransmissions left

Same with DISCOVER. Without DTLS I can clearly see the packets on Wireshark, with DTLS Wireshark is recognizing them as COAP but with "Unknown 254". My assumption is that's because of the encryption. However I can see one packet to the node and the answer from the node as well.

Gr3yh0und, 2017-03-28:
Did manage to get it working. Not sure what changed, but building the node firmware with `make TARGET=cc2538dk WITH_COAPSERVER=1 WITH_TINYDTLS=1  WITH_DTLS_COAP=1` and using `java -jar cf-client-1.1.0-SNAPSHOT.jar -psk DISCOVER coaps://[fd00::212:4b00:615:a86b]:5683/` works now...

closed 2017-03-28 by Gr3yh0und

laurentderu, 2017-03-31:
Thanks for your feedback, I will try to improve the documentation with your finding.

msolters, 2017-04-07:
Hey @Gr3yh0und, thank you for your work.  I have read your blog posts as well.

When implementing DTLS to secure CoAP communications on a mesh, I think I follow the gist of building the node firmware.  But there is one point on which I am still unclear -- how and/or where is DTLS implemented in the border router itself?

*  Is DTLS in any way configurable through 6LBR's web interface?
*  To use DTLS, must the 6LBR binary itself (e.g. `examples/6lbr`) have to be recompiled with DTLS compile flags?

Gr3yh0und, 2017-04-07:
Hey @msolters,
Yeah you are right, I haven't covered that part yet. I've already got a draft but was forced to first have a closer look into RIOT....

TO answer your questions:

1 & 2: DTLS is not part of the border router. This makes sense because the idea behind that is end-to-end encryption. Having the border router somehow handle/interprete/alter the content in between would be a high security risk (man in the middle).
DTLS is purely part of the application on the contiki leaf. In this case of the examples/6lbr-demo. If you check the [wiki](https://github.com/cetic/6lbr/wiki/Example-:-Dtls-Coap-Server) you can find that for the leaves you need to add the environment settings `WITH_TINYDTLS=1 WITH_COAPSERVER=1 WITH_DTLS_COAP=1` to include everything into the 6lbr-demo firmware.

3: No. As 1&2 point out you just have the coap server on the leaf which is using tinyDTLS as "exit" for the packets instead of using just a raw IPv6 socket.

Hope that helps and I will try to cover that in another blog entry as well.

msolters, 2017-04-07:
I see!  So:

*  The border router knows nothing about DTLS.
*  It is ultimately up the server and/or client on the other end to implement DTLS with a matching configuration.

This leads me to the next questions:

*  The leaf nodes can operate as CoAP servers and/or clients.  When compiling with tinyDTLS, do both of these modes then use DTLS?
*  If yes, do the server and client modes use the same configuration (e.g. the same certs or PSKs)?

I am going to be exploring this myself using the various DTLS NPM packages out there, but any guidance or knowledge you can share on this would be very appreciated!

Gr3yh0und, 2017-04-07:
Afaik both client and server use tinyDTLS and both also use the predefined/hardcoded standard tinyDTLS PSK which you find in the source code.