cetic/6lbr #282: Openwrt with 6lbr issue

knowic, 2017-03-29:
Hello, sir,

I have succeed with configuration of 6lbr on Openwrt platform, and get it start.
But I can get the NA message with LAN, so all the PC nodes in the same LAN can't get an IPV6 address like bbbb::xxxx, Can any one give some help about this? ( I have disable the firewall of openwrt), I pasted the interface information and 6lbr.log as follows:

My Openwrt network interface is as this:

root@OpenWrt:/etc/init.d# ifconfig
apcli0    Link encap:Ethernet  HWaddr 66:51:7E:00:29:5C
          inet6 addr: fe80::6451:7eff:fe00:295c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

br-lan    Link encap:Ethernet  HWaddr 64:51:7E:80:29:5C
          inet addr:192.168.8.1  Bcast:192.168.8.255  Mask:255.255.255.0
          inet6 addr: fe80::6651:7eff:fe80:295c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:1080 (1.0 KiB)

eth0      Link encap:Ethernet  HWaddr 64:51:7E:80:29:5C
          inet6 addr: fe80::6651:7eff:fe80:295c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6476 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1428 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:633067 (618.2 KiB)  TX bytes:492891 (481.3 KiB)
          Interrupt:5

eth0.1    Link encap:Ethernet  HWaddr 64:51:7E:80:29:5C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:44 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2695 (2.6 KiB)  TX bytes:2169 (2.1 KiB)

eth0.2    Link encap:Ethernet  HWaddr 64:51:7E:80:29:5D
          inet addr:10.10.12.176  Bcast:10.10.15.255  Mask:255.255.240.0
          inet6 addr: fe80::6651:7eff:fe80:295d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6432 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1401 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:513804 (501.7 KiB)  TX bytes:483352 (472.0 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:30 errors:0 dropped:0 overruns:0 frame:0
          TX packets:30 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2650 (2.5 KiB)  TX bytes:2650 (2.5 KiB)

ra0       Link encap:Ethernet  HWaddr 64:51:7E:80:29:5C
          inet6 addr: fe80::6651:7eff:fe80:295c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:6

tap0      Link encap:Ethernet  HWaddr 02:0A:0B:0C:0D:0E
          inet6 addr: bbbb::a:bff:fe0c:d0e/64 Scope:Global
[... 72 lines cut ...]
2017-03-29 22:04:31.485744: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:31.769849: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:31.770119: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:31.770279: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:31.770438: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:31.770595: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:38.190607: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:5a9:8fa5
2017-03-29 22:04:38.190804: INFO: 6LBR: Starting as RPL ROUTER
2017-03-29 22:04:38.191288: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-03-29 22:04:38.191406: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-03-29 22:04:38.191487: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2017-03-29 22:04:38.191601: INFO: DNS: DNS proxy started
2017-03-29 22:04:38.191681: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-03-29 22:04:38.191768: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-03-29 22:04:38.191840: INFO: 6LBR: CETIC 6LBR Started
RD client started
2017-03-29 22:04:40.534733: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535000: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535158: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535315: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535474: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535632: INFO: SLIP: uart_isr input fired
2017-03-29 22:04:40.535789: INFO: SLIP: uart_isr input fired

laurentderu, 2017-03-31:
Check if the multicast snooping workaround is properly activated on br-lan. As OpenWRT manages the interfaces directly, you should either add it into /etc/6lbr/6lbr.conf file using :

    BRIDGE_DISABLE_SNOOPING=1

or into /etc/sysctld

Another possibility would be that the tap interface is not bridged with br-lan, to do so you have to create a virtual interface in OpenWRT associated with tap0 and add it to br-lan bridge.

knowic, 2017-03-31:
**Hi,  
The previous  6lbr.log file show that the can't load nvm.conf , so I have edit the /usr/bin/6lbr script to load correct 6lbr.conf file,  my 6lbr.conf is copied from 6lbr.conf.openwrt as following:**

#This file contains a default configuration for OpenWRT platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example
MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br-lan
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB0
BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only

#6LBR Watchdog does not work yet on OpenWRT
WATCHDOG_INTERVAL=0

#OpenWRT kernel still has multicast snooping bug
BRIDGE_DISABLE_SNOOPING=1

**But when I restart the 6lbr( /etc/init.d/6lbr restart ), I got such an syntax error reports, could you help me?
Thanks for your attention and appreciate your help!**

31 22:13:29 CST 2017 : Starting 6LBR
/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/6lbr.conf -s /dev/ttyUSB0 -R -t tap0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /u
3.0 started with IPV6, RPL
rted with address 1.2.3.4.5.6.7.8
wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
31 22:13:29.798630: NOTICE: 6LBR: Starting 6LBR version 1.4.1 (Contiki 3.0)
31 22:13:29.798968: WARN: 6LBR: 6LBR Watchdog disabled
31 22:13:29.799380: INFO: NVM: NVM Magic : 2009
31 22:13:29.799477: INFO: NVM: NVM Version : 5
31 22:13:29.799548: INFO: CONFIG: Loading configuration : /etc/6lbr/6lbr.conf
31 22:13:29.799848: FATAL: CONFIG: Syntax error in /etc/6lbr/6lbr.conf at line 4
************************************************************************
31 22:13:30 CST 2017 : Starting 6LBR
/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/6lbr.conf -s /dev/ttyUSB0 -R -t tap0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /u
3.0 started with IPV6, RPL
rted with address 1.2.3.4.5.6.7.8
wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
31 22:13:30.825802: NOTICE: 6LBR: Starting 6LBR version 1.4.1 (Contiki 3.0)
31 22:13:30.826114: WARN: 6LBR: 6LBR Watchdog disabled
31 22:13:30.826556: INFO: NVM: NVM Magic : 2009
31 22:13:30.826649: INFO: NVM: NVM Version : 5
31 22:13:30.826719: INFO: CONFIG: Loading configuration : /etc/6lbr/6lbr.conf
31 22:13:30.827007: FATAL: CONFIG: Syntax error in /etc/6lbr/6lbr.conf at line 4

hellosfs, 2017-04-10:
hello ,sir~
can you visit your bbbb::100 on openwrt?
such as wget http://[bbbb::100]

laurentderu, 2017-04-11:
6lbr.conf and nvm.conf are two different files and have completely different content and you can not swap them.

The former contains the basic configuration of 6LBR. the latter contains more advanced or dynamic configuration, by default it does not exist or is empty.

closed 2017-06-15 by laurentderu