cetic/6lbr #283: 6lbr as smart-bridge with static ipv6 configuration

fsismondi, 2017-03-29:
Hello all,
I've been having problems to configure 6LBR as a bridge with an static ipv6 configuration into a RPi, basically what I cannot get running is to be able to access http://[bbbb::100] from my computer. 

I setup my PC eth interface w/ IPv6 (manually) of my PC without problems. But I can see in wireshark in that same interface plenty of DHCP requests and ICMPv6 Nighbor solicitations, which makes me think that I misconfigured sth in the LBR router side.

I'm following this:
https://github.com/cetic/6lbr/wiki/6LBR-Connection
https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

I would like an example if possible for a RPi on this:

Manual connection

If you don’t want to have a running router advertisement daemon, you can manually configure the prefix in 6LBR in line with the used prefix on the Ethernet segment.

Thanks in advance.

laurentderu, 2017-03-31:
If the BR is correctly configured in Router mode and bridged properly with the Eth interface you should see periodical RA sent by the RPi. If not it could be one of the following causes :

* 6LBR is not configured in Router mode
* The tap0 interface is not bridged with the br0 interface or a network manager has reconfigured the bridge
* The multicast snooping workaround is not activated on the bridge, either add it in your 6lbr.conf file or /etc/network/interfaces
* You are using WiFi to connect to the RPi It can only work if the WiFi on the RPi is configured as AP.

fsismondi, 2017-03-31:
First, thanks for the answer.

Now, if I understood correctly, your suggestions are for setting up a 6LBR as a standard border router, but what I'm trying to achieve is a 6LBR setup in smart-bridge mode with manual configuration of the prefix for the 6LBR. 

My problem is that I couldn't setup that prefix, do you have an example on how to do this for RPi?

laurentderu, 2017-04-11:
Sorry, I misunderstood your setup. But almost all the causes mentioned above are also applicable in case of SmartBridge mode :)

Depending on the version of 6LBR, the default configuration is different though :
On 1.4.x and earlier, by default 6LBR waits for a RA message to auto configure itself
On 1.5.x and later, by default 6LBR auto configures itself with the default WSN prefix fd00::/64

To modify this behavior without having access to the web server of the BR, you can use the nvm_tool to update the configuration (when 6LBR is stopped) :

    /usr/lib/6lbr/bin/nvm_tool --update --wsn-accept-ra 0 --wsn-prefix abcd:: /etc/6lbr/nvm.dat

The IP address of the BR will be abcd::<BR radio IID>

closed 2017-12-14 by laurentderu