cetic/6lbr #285: 6lbr webserver lite

MarCast, 2017-04-03:
Hi everyone, I managed to connect my CC1310 5X5 with the ENC28J60 and i can ping [bbbb::100], the result is the 6lbr webserver-lite. Now I would like to connect a node to monitor how the CC26xx-web-demo. Is there a way?

MarCast, 2017-04-03:
@laurentderu @g-oikonomou

chenek, 2017-04-05:
I test to run  cc26xx-web-demo to join my LAUNCHXL-CC1310+ENC28J60 runing 6lbr Lite and it works without problem. The only revision I need to do is to change "#define RF_CORE_CONF_CHANNEL 25" to "#define RF_CORE_CONF_CHANNEL 26" in project-conf.h when I build cc26xx-web-demo.

MarCast, 2017-04-06:
Where the genres of the file? Under Contiki or 6lbr?

chenek, 2017-04-06:
I couldn't understand your question. genres of the file?

closed 2017-04-06 by MarCast