cetic/6lbr #287: 6lbr to external network

MarCast, 2017-04-06:
Hi everyone, I would like to connect the 6lbr from "external network". I used a port forwarding but is not the solution that is right for me. Is there another way? There is some wires of the configuration to be modified? Thanks of all. @laurentderu @g-oikonomou

MarCast, 2017-04-07:
@JelmerT can you help me?

laurentderu, 2017-04-11:
If you have a public IPv6 subnet, you can use native IPv6 (and set up your firewall to protect the BR), or otherwise you can enable NAT64 in 6LBR or use an external NAT64 service. You can have more information about NAT64 here : https://github.com/cetic/6lbr/wiki/NAT64

whu-inf, 2017-04-12:
@laurentderu Hi. My pc managed to reach [bbbb::100] now, and found that it's a 'lite' web page with no options to configure IP64. My 6lbr is running on a srf06/cc1310. What should I do to get IP64?
The reason why I wanna try IP64 is that my Android phone can NOT reach [bbbb::100] using WLAN, while my Windows PC can. My PC is wired to the LAN port of the openwrt router. I noticed, from the router's webconfig page, that 'lan' and 'wlan' are bridged through a 'br_lan'. And my phone got a local ipv6 address like fe80::xxxx and an ipv4 addr. Should I make some modifications in openwrt to fix this 'android-down' or give IP64 a try?

laurentderu, 2017-04-18:
IP64 can either be enabled using the NVM-Proxy (see https://github.com/cetic/6lbr/wiki/Nvm-Proxy), which is a small tool to upload a nvm configuration into the BR, or you can modify the default values of the NVM in nvm-data.h (in the next release you should be able to modify them directly in your project-conf.h file).

MarCast, 2017-05-10:
Yes, but I wouldn't like the NAT64 and Port Forwarding.

closed 2017-10-12 by MarCast