cetic/6lbr #288: Sensor Export default disable

steven3w, 2017-04-07:
Hi 6lbr team,
I notice the V1.4.1 has Export added at Sensors tab on the 6lbr web page. I have tested it and found the default is disable. How to make it default enable. And is there way I can limit the record number?

laurentderu, 2017-04-11:
This feature is still experimental and to be documented :)

You configure it using the /etc/6lbr/nvm.conf file with the following entry :

    [node-info.export]
    enabled = 0 | 1 #Enable or disable the functionality at BR startup
    global= 0 | 1 #Dump date in one global file or a file per node
    filename=... #Name of the dump file if global file is used
    path=... #Path of the dump subdirectory 
    interval=   #Interval between dumps in seconds

Currently there is no limit, but indeed this could be added. What kind of limit would you suggest, time ? number of records ?

steven3w, 2017-04-12:
hi, Laurentderu,

Thank you. I check my beagle bone black, there is no /etc/6lbr/nvm.conf file there. Do you mean I can create a new file with the option contents?

As for the limit, number limit is better, I think, because if I access it from a application program I want read it as fast as possible.

Steven Dong

steven3w, 2017-04-13:
Hi, Laurentdeu, 

I suggest export should include RSSI. It is useful to express a link's quality.

Steven

Dimaqa, 2019-02-21:
> enabled = 0 | 1 #Enable or disable the functionality at BR startup

It should be `enable=1`, i believe.