cetic/6lbr #290: How to make MQTT message sent to IBM Bluemix using 6lbr Lite running on LAUNCHXL-CC2650

chenek, 2017-04-07:
I can run 6lbr Lite on LAUNCHXL-CC2650 + ENC28J60 and join another  LAUNCHXL-CC2650 running cc26xx-web-demo. I can access to sensor web page and coap server on cc26xx-web-demo. However, I cannot see MQTT messages sent to IBM Bluemix. I believe it's because 6lbr Lite doesn't include NAT64 capability. So, how to make MQTT message sent to IBM Bluemix using 6lbr Lite running on LAUNCHXL-CC2650?

paishreeni, 2017-04-07:
I'm using srf06-cc2650. even I've the same issues.

whu-inf, 2017-04-12:
Similar issue. And btw have you ever tried to access sensor webpage through WLAN, like,  using an android phone? My LAN-wired-PC has access to those webpages but my cellphone doesn't.

closed 2017-12-14 by laurentderu