cetic/6lbr #291: 6lbr Lite issue on LAUNCHXL-CC2650 + ENC28J60

chenek, 2017-04-10:
I can run 6lbr Lite on LAUNCHXL-CC2650 + ENC28J60 (I see UART output and can connect to bbbb::100) first time after I download cetic_6lbr_router.bin to LAUNCHXL-CC2650. However, if I press the reset button on LAUNCHXL-CC2650 or power recycle LAUNCHXL-CC2650, I cannot see 6lbr Lite restarts on LAUNCHXL-CC2650 + ENC28J60. There is no UART output and I cannot connect to bbbb::100. I also test this no srf06-cc2650EM+ENC28J60 and result is the same. Since there is nothing output from UART, it's difficult to debug what happens. Does anyone know how to fix this?

tarakanov, 2017-04-10:
I have the same issure with sensortag. It seems that CC2650 have less internal NVM memory, so you need to use external flash to store NVM data. Try to make with `CC26XX_NVM=ext`

chenek, 2017-04-10:
I tried to build using "make CC26XX_NVM=ext TARGET=srf06-cc26xx BOARD=srf06/cc26xx all" but it pops the following link errors:

obj_srf06-cc26xx/cc26xx-ext-nvm.o: In function `nvm_data_read':
cc26xx-ext-nvm.c:(.text.nvm_data_read+0xe): undefined reference to `ext_flash_open'
cc26xx-ext-nvm.c:(.text.nvm_data_read+0x2e): undefined reference to `ext_flash_read'
cc26xx-ext-nvm.c:(.text.nvm_data_read+0x34): undefined reference to `ext_flash_close'
cc26xx-ext-nvm.c:(.text.nvm_data_read+0x24): undefined reference to `ext_flash_close'
obj_srf06-cc26xx/cc26xx-ext-nvm.o: In function `nvm_data_write':
cc26xx-ext-nvm.c:(.text.nvm_data_write+0xe): undefined reference to `ext_flash_open'
cc26xx-ext-nvm.c:(.text.nvm_data_write+0x24): undefined reference to `ext_flash_erase'
cc26xx-ext-nvm.c:(.text.nvm_data_write+0x3a): undefined reference to `ext_flash_write'
cc26xx-ext-nvm.c:(.text.nvm_data_write+0x50): undefined reference to `ext_flash_close'
collect2: error: ld returned 1 exit status
make[2]: *** [cetic-6lbr.elf] Error 1

I think all those ext_flash_xxx are missed in CC26xx 6lbr. How can I fix this?

chenek, 2017-04-10:
Yes, changing the macro CETIC_6LBR_NVM_SIZE in file cc26xx-int-nvm.c fixes the issue using internal NV. But, what if I would like to use external flash, how can I fix the build problem?

mcuxmx, 2017-04-10:
the board srf06/cc26xx with no  external flash
if you want use it, you should change the relevant source files and makefile files

chenek, 2017-04-10:
Thanks for the hint. I fix the issue by using "make TARGET=srf06-cc26xx BOARD=launchpad/cc2650 all"

closed 2017-04-10 by chenek