cetic/6lbr #293: Promiscuous mode of BBB

steven3w, 2017-04-12:
Hi, Laurentderu,
In the wiki, interface configuration you have the following statements:

Important note: Your Ethernet driver must support Promiscuous mode! If not the bridge won't work properly and you will have intermittent connectivity with 6LBR. The Ethernet driver of the Beaglebone does not support promiscuous mode!

I am using BeagleBone Black, I have set it to router with bridge, and write a nodejs app to test with udp communication with the contiki node at about 50ms each send and receive, It work with no lost packet for a couple of days. 

 Is the statement in the wiki outdate or it is only for Beaglebone, but nor for Beaglebone Black. I install Jessie 7.9.

Steven Dong

laurentderu, 2017-04-18:
The statement is applicable for both Beaglebone and Beaglebone Black. But it may be possible that bridging support has been added to the kernel driver in later kernel versions.
You should check in the kernel log for warnings about promiscuous mode not supported. If none are present, that may indicate that eventually promiscuous support has been added.

steven3w, 2017-04-19:
Hi, Laurentderu,

I have checked the log:

debian@beaglebone:~$ dmesg -T | grep eth0
[Sat Apr  8 17:20:50 2017] net eth0: initializing cpsw version 1.12 (0)
[Sat Apr  8 17:20:50 2017] net eth0: phy found : id is : 0x7c0f1
[Sat Apr  8 17:20:50 2017] net eth0: phy 4a101000.mdio:01 not found on slave 1
[Sat Apr  8 17:20:50 2017] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[Sat Apr  8 17:20:53 2017] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[Sat Apr  8 17:20:56 2017] device eth0 entered promiscuous mode
[Sat Apr  8 17:20:56 2017] net eth0: promiscuity ignored!
[Sat Apr  8 17:20:56 2017] net eth0: Ignoring Promiscuous mode
[Sat Apr  8 17:20:56 2017] net eth0: Ignoring Promiscuous mode
[Sat Apr  8 17:20:56 2017] net eth0: Ignoring Promiscuous mode
[Sat Apr  8 17:20:56 2017] br0: port 1(eth0) entered forwarding state
[Sat Apr  8 17:20:56 2017] br0: port 1(eth0) entered forwarding state
[Sat Apr  8 17:21:02 2017] net eth0: Ignoring Promiscuous mode
[Sat Apr  8 17:21:12 2017] br0: port 1(eth0) entered forwarding state     

It seems the BBB indeed report error on promiscuous mode.

However, the system works fine.

I have just done a 6 days test with one contiki tag receiving a UDP from BBB through the 6lbr, and send the packet back, and after BBB receive the sent back packet I start it again, I have totally achieved 800,000 times send-receive without any errors, and this takes about 6 days.

So I really doubt the statement in the wiki. 

cheers
Steven Dong

laurentderu, 2017-04-19:
I will give a try, but beware that the promiscuous bug is a tricky one as your system seems to work but some packets are never forwarded from the Eth0 interface to the bridge. Having a frequent traffic usually hides the problem, but from time to time you will experiment loss of connection towards the BR.

closed 2017-11-09 by laurentderu

zengxu-yang, 2018-10-29:
I checked the Linux kernel source code. CPSW promiscuous mode was added in Linux kernel 3.14. I use the new Debian Stretch on BBB and it seems that the promiscuous mode is working with newer kernels. No error messages from dmesg.