cetic/6lbr #295: issue at very 1st step ...

sagarbhanushali, 2017-04-18:
DTLS echo server example:
issue at very 1st step .... not able to build files.
steps followed:
1. git cloned repo
2. configured tiny dtls in apps.
3. Using this command to build files:
- make TARGET=wismote WITH_TINYDTLS=1 WITH_DTLS_ECHO=1
lots of errors are generated....can someone suggest a stepwise approach for this. 
Thanks in advance

laurentderu, 2017-04-18:
Could you provide the error log ? Be aware that building for wismote (or any MSP430x) requires mspgcc 4.7.2

sagarbhanushali, 2017-04-18:
hello,
Thank you for replying me,
Hope you are doing good.

I reinstalled everything and now able to build the files and used this
command:

cd examples/6lbr-demo
make TARGET=wismote WITH_TINYDTLS=1 WITH_COAPSERVER=1 WITH_DTLS_COAP=1

now what is the procedure to run this in cooja simulator
as if I use wismote and feed 6lbr-demo.c and compile then new object will
be formed right?
correct me if I am wrong.

Also, in installing maven, it is showing unable to locate package maven.
I used this link to install maven
https://www.mkyong.com/maven/how-to-install-maven-in-ubuntu/ .
Please suggest me what to do.... as I want to integrate coap with dtls over
6lowpan and plese reply as soon as possible as it's very urgent.

Thanks and Regards,

Sagar  Bhanushali
sagarbhanushali1991@gmail.com
www.sagarbhanushali.com
+1 9792679842

laurentderu, 2017-04-19:
You can build in cog by creating a wismote node (or exp5438) and use 6lbr-demo.c with the proper compilation flags as source for the firmware. However you will need to run an instance of 6LBR and connect it to the simulation using a slip-radio.
For that you can look at the documentation : https://github.com/cetic/6lbr/wiki/COOJA-Interface
 (Beware that the default simulation uses coojamote which are not compatible with Wismote or sky)

sagarbhanushali, 2017-04-19:
Thanks for replying me.

1. I have successfully compiled and created wismote node using 6lbr-demo.c
with the proper compilation flag and simulated in cooja simulator.
now will need a coaps client to interact with the server. correct me if I
am going on the wrong path.
2. For this, I was trying to configure californium client but getting
issues in installing maven.
- following errors are faced during installation:
 - using sudi apt-get install maven but getting no package found .... tried
this link https://www.mkyong.com/maven/how-to-install-maven-in-ubuntu/
Please suggest me.
Thankyou in advance.

Thanks and Regards,

Sagar  Bhanushali
sagarbhanushali1991@gmail.com
www.sagarbhanushali.com
+1 9792679842

sagarbhanushali, 2017-04-19:
Thanks for replying me.

1. I have successfully compiled and created wismote node using 6lbr-demo.c with the proper compilation flag and simulated in cooja simulator.
now will need a coaps client to interact with the server. correct me if I am going on the wrong path.
2. For this, I was trying to configure californium client but getting issues in installing maven.
- following errors are faced during installation:
 - using sudo apt-get install maven but getting no package found .... tried this link https://www.mkyong.com/maven/how-to-install-maven-in-ubuntu/ 
Please suggest me.
Thank you in advance.

Thanks and Regards,

Sagar  Bhanushali
sagarbhanushali1991@gmail.com
www.sagarbhanushali.com
+1 9792679842

laurentderu, 2017-04-19:
I'm sorry I can't help you with maven or Californium specific errors, you should check the Californium documentation and ask their support  mailing list.

For testing purposes, I instead use Copper plugin for Firefox, it's much easier to setup and use to test your server than building a custom Californium client, unless you need to test DTLS.

sagarbhanushali, 2017-04-19:
Thanks for replying.
But if I want see the handshaking between dtls client and server so how i
can I chwck using coppen plugin in firefox.
- i have already installed copper plugin and used for coap servers but how
to use for coaps server i dont know....
please suggest.

Thanks in advance.

closed 2017-11-09 by laurentderu