cetic/6lbr #300: Problem with dwc_otg.speed=1 on raspberry

NicolasLeGall, 2017-05-11:
I use Raspbian GNU/Linux 8.0 (jessie) on a raspberry. I was following this guide to configure my raspberry: https://github.com/cetic/6lbr/wiki/RaspberryPi-Software-Configuration
When I modify /boot/cmdline.txt and write the line dwc_otg.speed=1 if I reboot my raspberry my keyboard doesn't work.
I realized that if I enter the bios of the RPi and I change the confiuration , the keyboards works on startup.
Is this a bug or a feature ? :slightly_smiling_face:

laurentderu, 2017-05-24:
It was a feature, now it might be a bug :)

In the past the RPi had problems with serial devices with 115200 kbps, like a slip-radio, and the parameter was required. Now it might be obsolete, I will update the documentation to suggest to add it only if the slip-radio does not work.

laurentderu, 2017-06-20:
Doc updated.

closed 2017-06-20 by laurentderu