cetic/6lbr #301: Example : DTLS CoAP Server, wismote compile error

Han-sangwoo, 2017-05-12:
I am testing Example : DTLS CoAP Server for cooja simulator.
But, 
> cd examples/6lbr-demo
> make TARGET=wismote WITH_TINYDTLS=1 WITH_COAPSERVER=1 WITH_DTLS_COAP=1
is compile failed.

Error Message is "/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: region `rom' overflowed by 2 bytes". and in case of TARGET=cc2538dk, compile success and create '6lbr-demo.bin','6lbr-demo.cc2538dk' but cooja simulator is not support cc2538dk.
so I tried Target = wismote, but failed. How I can compile TARGET=wismote?

tuncagurkan, 2018-11-25:
@Han-sangwoo , Did you fix the issue? If yes, please share with us. I got same error.

tuncagurkan, 2018-12-24:
I fixed this issue, updating msp430 is enough.

SalvaHasan, 2021-09-24:
I'm getting the same overflow error on cc2560 SensorTag. Any idea what we can disable within the project-conf.h file to save space and fit it in?