cetic/6lbr #307: unable to see TI sensortag 2.0 from webpage

naivewong, 2017-06-07:
According to the link http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples i used beaglebone black, CC2531 USB dongle and CC2650 sensorTag. And i succeeded in accessing the website [bbbb::100] and i was able to see the sensorTag on [bbbb::100]/sensors.html.
 
Then i replaced the beaqglebone black with the raspberry pi and my raspberry’s version was 4.4.50-v7+. And after configuration i was able to access [bbbb::100] but i was not able to see the sensorTag on the website. I found a link https://github.com/cetic/6lbr/issues/92 on github and its problem was the same with me. But the version of his raspberry pi was older. I didn’t have file /etc/inittab and the content of /boot/cmdline.txt was different. Now i have no idea how to configure it. There are several relevant pictures.
Thank you for your attention!
    
   
[image]

 
[image]

[image]

[image]
[image]
[image]
[image]
[image]
[image]

laurentderu, 2017-06-07:
If you remove console=serial0, 115200 from your cmdline.txt does it solve the issue ? Also, try with removing dwc_otg.speed=1

naivewong, 2017-06-08:
I first removed console=serial0, 115200 and it didn't work. Then I removed dwc_otg.speed=1 and it still didn't work. Is the problem with Raspberry's version or 6lbr's version?

naivewong, 2017-06-08:
@laurentderu I first removed console=serial0, 115200 and it didn't work. Then I removed dwc_otg.speed=1 and it still didn't work. Is the problem with Raspberry's version or 6lbr's version?

laurentderu, 2017-06-20:
I forgot that there is another daemon span by the system to provide a console on the UART, you must disable it either using raspi-config or manually, see https://github.com/cetic/6lbr/wiki/RaspberryPi-Software-Configuration#specific-configuration-for-the-uart-based-radio

closed 2017-11-09 by laurentderu

Usman78930, 2019-05-18:
I am also facing the same issue. i am using raspbian stretch. Kindly help me fix this problem