cetic/6lbr #308: 6lbr is not showing webpage when working with BBB

rahultt007, 2017-06-08:
Hi,

We are trying to create a 6LoWPAN Mesh network using CC1350 Launchpads as End Devices and CC1350 Launchpad connected to BBB as Edge router (which needs to convert IPV6 to IPV4). We are planning to send our sensor readings from End Device to server via MQTT Protocol. We did the following steps,

1) Downloaded and installed Contiki in Virtual box.
2) Cloned the latest repo of contiki as per this link.
http://sunmaysky.blogspot.in/2015/09/contiki-subg-hz-6lowpan-on-cc1350.html?m=1
3) Created bin images for web demo (cmd = make TARGET=srf06-cc26xx BOARD=launchpad/cc1350 cc26xx-web-demo.bin) and slip radio (make TARGET=srf06-cc26xx BOARD=launchpad/cc1350 slip-radio.bin).
We followed this link for configuring BBB. http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples
BBB is connected to PC via USB interface and both BBB and PC is in same network. But we are not able to get the web page in browser.

here is the 6lbr.log

_********************************************************************************
Thu Jun 8 11:09:21 UTC 2017 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-develop-20161222 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2017-06-08 11:09:21.889991: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-develop-20161222)
2017-06-08 11:09:21.891030: INFO: ETH: 6LBR watchdog started (interval: 60)
2017-06-08 11:09:21.902700: INFO: NVM: NVM Magic : 2009
2017-06-08 11:09:21.902748: INFO: NVM: NVM Version : 5
2017-06-08 11:09:21.902760: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-06-08 11:09:21.902862: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-06-08 11:09:21.903076: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2017-06-08 11:09:21.903095: INFO: MAC: Using 'csma' mac driver
2017-06-08 11:09:21.905220: INFO: SLIP: SLIP started on /dev/ttyACM0
2017-06-08 11:09:21.915986: INFO: SCMD: Started br-cmd process
2017-06-08 11:09:21.916127: INFO: BR-RDC: Reset SLIP Radio
2017-06-08 11:09:21.916172: INFO: BR-RDC: Fetching MAC address
2017-06-08 11:09:21.919557: ERROR: SLIP: Unknown command
2017-06-08 11:09:21.920367: INFO: BR-RDC: Got MAC: 00:12:4b:00:0e:00:b5:01
2017-06-08 11:09:22.918491: ERROR: SLIP: Unknown command
2017-06-08 11:09:22.918861: ERROR: BR-RDC: Set channel failed
2017-06-08 11:09:22.921810: ERROR: SLIP: Unknown command
2017-06-08 11:09:22.922038: ERROR: BR-RDC: Set PAN-ID failed
2017-06-08 11:09:22.922175: INFO: ETH: RAW Ethernet interface init
2017-06-08 11:09:22.938277: INFO: TAP: opened device /dev/eth0
2017-06-08 11:09:22.938675: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b0:d5:cc:fc:3a:8f
          inet addr:192.168.3.12  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: fe80::b2d5:ccff:fefc:3a8f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:556 errors:0 dropped:1 overruns:0 frame:0
          TX packets:230 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:71285 (69.6 KiB)  TX bytes:34108 (33.3 KiB)
          Interrupt:40

2017-06-08 11:09:23.103811: INFO: TAP: Eth MAC address : b0:d5:cc:fc:3a:8f
2017-06-08 11:09:23.104270: INFO: ETH: Eth MAC address : b0:d5:cc:fc:3a:8f
2017-06-08 11:09:23.104439: INFO: LLSEC: Using 802.15.4 framer
2017-06-08 11:09:23.104545: INFO: LLSEC: Using 'nullsec' llsec driver
2017-06-08 11:09:23.105119: INFO: HTTP: Starting webserver on port 80
2017-06-08 11:09:23.105815: INFO: NODECFG: Node Config init
2017-06-08 11:09:23.105982: INFO: NODECFG: No node_config.conf file specified
2017-06-08 11:09:23.106325: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e00:b501
2017-06-08 11:09:23.106656: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:e00:b501
2017-06-08 11:09:23.106970: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2017-06-08 11:09:23.107163: INFO: 6LBR: RA Daemon enabled
2017-06-08 11:09:23.107261: INFO: 6LBR: Checking addresses duplication
2017-06-08 11:09:34.109545: INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:e00:b501
2017-06-08 11:09:34.109829: INFO: 6LBR: Starting as RPL ROUTER
2017-06-08 11:09:34.110989: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-06-08 11:09:34.111104: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-06-08 11:09:34.111188: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2017-06-08 11:09:34.111423: INFO: DNS: DNS proxy started
2017-06-08 11:09:34.111491: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2017-06-08 11:09:34.111559: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2017-06-08 11:09:34.111621: INFO: 6LBR: CETIC 6LBR Started
RD client started_

debian@beaglebone:~$ /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'
Channel : 25
PAN ID : abcd

WSN network prefix : fd00::
WSN network prefix length : 64
WSN IP address : fd00::100
WSN accept RA : False
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : fd00::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 4
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

RPL DAO ACK : True
RPL DAO ACK local repair : False
DIO triggers route refresh : False
MAC layer : 1
Security layer : 0
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled : False
Noncoresec anti-replay workaround : False
Filter unknown nodes : False

NAT 64 Enabled : False
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : False
NAT 64 RFC-6052 prefix : False
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80

debian@beaglebone:~$

Please help us in solving the issue.

Thanks
Rahul

laurentderu, 2017-06-15:
The problem is that the BR interface, with IP bbbb::100, is not exposed on the USB interface but on the Ethernet interface (as 6LBR is running in Raw Ethernet mode). So you need to connect through the Ethernet interface and configure your PC to either accept interface autoconfiguration or manually configure an address in the bbb::/64 range.

rahultt007, 2017-06-20:
Thanks for your response. Now we are able to get the webpage. But when we run the 6lbr service we are not getting internet in BBB and we will get it once we stop the service. Sometimes needs to start and stop the 6lbr to get internet in BBB. Please let me know the issue.

Our **6br.config** is as follows,

_#This file contains a default configuration for BeagleBone platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=6LBR
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

#Enable this line if radio device is not detected
#SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

LOG_LEVEL=3 #INFO and above only_

And our interfaces file is as follows,

_# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# The loopback network interface
auto lo
iface lo inet loopback

#iface eth0 inet static
#address 0.0.0.0

#auto br0
#iface br0 inet dhcp
# bridge_ports eth0
# bridge_stp off
# up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
# post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`

# The primary network interface
auto br0
iface br0 inet dhcp
#auto eth0
#iface eth0 inet dhcp
# Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

##connman: ethX static config
#connmanctl services
#Using the appropriate ethernet service, tell connman to setup a static IP address for that service:
#sudo connmanctl config <service> --ipv4 manual <ip_addr> <netmask> <gateway> --nameservers <dns_server>

##connman: WiFi
#
#connmanctl
#connmanctl> tether wifi off
#connmanctl> enable wifi
#connmanctl> scan wifi
#connmanctl> services
#connmanctl> agent on
#connmanctl> connect wifi_*_managed_psk
#connmanctl> quit

# Ethernet/RNDIS gadget (g_ether)
# Used by: /opt/scripts/boot/autoconfigure_usb0.sh
iface usb0 inet static
address 192.168.7.2
netmask 255.255.255.252
network 192.168.7.0
gateway 192.168.7.1_

Please let me know your thoughts on this.

Thanks

Rahul

akib-islam, 2017-08-29:
Hi,

I am getting the same 6lbr.log file as you attached above. I am not able to access the webpage. I can see from the above comments that you have managed to display the webpage. Can you tell me the steps for displaying the webpage.

Thanks,
Akib