cetic/6lbr #311: CC2650STK: Disconnect from 6lbr occasionally

naivewong, 2017-06-21:
I used raspberry pi with rasbian 4.1 running 6lbr whose version was 1.3.3. And i used nullRDC on cc2650stk. At fist, i can access the [bbbb::100]/sensors.html and the graph of the two sensortags is perfect. But after a while when i clicked the "web", randomly one sensortag may be disconnected and it would not reconnect. Just like the screenshot below. And then i switched to Beaglebone whose 6lbr version was also 1.3.3 and the same disconnection problem occurred.
[image]

laurentderu, 2017-06-21:
I believe this is a bug solved in the later 1.4.x versions. Could you check with version 1.4.1 or one of the dev snapshots ?

naivewong, 2017-06-24:
@laurentderu I tried this morning. And I found that at first I can access the "web" of two sensors, but after a while, one of the sensors' "web" can not be accessed. But strangely the connection showed on the webpage of 6lbr is complete. Just like this two pics.
[image]
[image]

closed 2017-11-09 by laurentderu