cetic/6lbr #312: Cannot access one of the "web" page of sensortag after starting my python web crwal program for a while

naivewong, 2017-06-25:
In order to get the sensors' data automatically, I wrote a python web crawler program to crawl these two "web" page of my two sensors. The program will access one sensor's 'web', crawl the data and store them in MySQL, then after 60 seconds, the program will change to another sensor. Although at first everything was fine, after a while I cannot access one of the sensor's web, sometimes 6387, sometimes a500... So I want to ask if there's any restriction on the accessing times of the 'web' of the sensors?
[image]
[image]
[image]

naivewong, 2017-06-25:
Or if there are some other ways to continually get the data?
Thank you.

laurentderu, 2017-11-09:
Solved in 5b43e1011bf8b7b6039a5b1eb109c6b2b730845e

closed 2017-11-09 by laurentderu