cetic/6lbr #318: Unable to get the 6lbr webpage

akib-islam, 2017-08-29:
Hi,

I am using Beabgle bone(as edge router) and cc1310 launchpad (as a slip radio). I have followed all the steps mentioned in 6lbr linux configuration page to configure my bbb as edge router. However I am not able to ping the ipv6 address bbbb::000, nvm.dat file is not created during run time and I am unable to get the webpage. Can anyone help me out to solve this issue. 

This is the content of 6lbr.conf 
`#This file contains a default configuration for BeagleBone platform using
#a Telos SLIP Radio
#The full list of parameters and their meaning can be found in 6lbr.conf.example

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

CREATE_BRIDGE=0

RAW_ETH=0
BRIDGE=1
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

ETC_6LBR=/etc/6lbr
NVM=/etc/6lbr/nvm.dat
LIB_6LBR=/usr/lib/6lbr
BIN_6LBR=$LIB_6LBR/bin
#IFDOWN=$LIB_6LBR/6lbr-ifdown
#IFUP=$LIB_6LBR/6lbr-ifup

DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200

#Enable this line if radio device is not detected
SCAN_DEVICE=/usr/lib/6lbr/bb-scan-usb.sh

#LOG_LEVEL=7 # Debug traces at packet level
#LOG_LEVEL=6 # Actual packet sent/received
#LOG_LEVEL=5 # Trace printed when a packet is sent/received
#LOG_LEVEL=4 # Debug information
LOG_LEVEL=3 # Runtime information
#LOG_LEVEL=2 # Unexpected condition occurred or important information message
#LOG_LEVEL=1 # Error detected and handled by 6LBR
`

This is my /etc/network/interfaces file:

`# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# The loopback network interface
auto lo
iface lo inet loopback

iface eth0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports eth0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`

# Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

# The secondary network interface
#auto eth1
#iface eth1 inet dhcp

# WiFi Example
#auto wlan0
#iface wlan0 inet dhcp
#    wpa-ssid "essid"
#    wpa-psk  "password"

# Ethernet/RNDIS gadget (g_ether)
# Used by: /opt/scripts/boot/autoconfigure_usb0.sh
iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.252
    network 192.168.7.0
    gateway 192.168.7.1
`
When I am starting the 6lbr, its getting started. Following is the output from 6lbr.log file

`Thu Nov 12 19:00:27 UTC 2015 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins
Contiki-contiki-base-develop-20170120-2375-gcfbc964 started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC multi-radio RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
2015-11-12 19:00:27.949477: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-base-develop-20170120-2375-gcfbc964)
2015-11-12 19:00:27.950283: INFO: SLIP: Allocated slip device 0 -> 0
2015-11-12 19:00:27.950480: INFO: ETH: 6LBR watchdog started (interval: 60)
2015-11-12 19:00:27.950874: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
2015-11-12 19:00:27.968308: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
2015-11-12 19:00:27.971901: INFO: PLUGIN: Initialising lwm2m-client
2015-11-12 19:00:27.972243: INFO: NVM: NVM Magic : 2009
2015-11-12 19:00:27.972300: INFO: NVM: NVM Version : 5
2015-11-12 19:00:27.972328: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2015-11-12 19:00:27.972382: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2015-11-12 19:00:27.972502: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2015-11-12 19:00:27.972533: INFO: MAC: Using 'csma' mac driver
2015-11-12 19:00:27.988676: INFO: SLIP: SLIP started on /dev/ttyACM0
2015-11-12 19:00:27.999343: INFO: SCMD: Started br-cmd process
2015-11-12 19:00:27.999518: INFO: BR-RDC: SLIP RADIO configured as RADIO
2015-11-12 19:00:27.999675: INFO: BR-RDC: Reset SLIP Radio
2015-11-12 19:00:27.999728: INFO: BR-RDC: Fetching MAC address
2015-11-12 19:00:28.010148: ERROR: SLIP: Unknown command
2015-11-12 19:00:28.010426: INFO: BR-RDC: Got MAC 0 : 00:12:4b:00:0e:07:c5:c3
2015-11-12 19:00:29.001962: ERROR: SLIP: Unknown command
2015-11-12 19:00:29.002285: ERROR: BR-RDC: Set channel failed
2015-11-12 19:00:29.005168: ERROR: SLIP: Unknown command
2015-11-12 19:00:29.005341: ERROR: BR-RDC: Set PAN-ID failed
2015-11-12 19:00:29.005416: INFO: ETH: RAW Ethernet interface init
2015-11-12 19:00:29.028143: INFO: TAP: opened device /dev/eth0
2015-11-12 19:00:29.028408: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
eth0      Link encap:Ethernet  HWaddr b0:d5:cc:f1:a9:60
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:40 errors:0 dropped:2 overruns:0 frame:0
          TX packets:18 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3672 (3.5 KiB)  TX bytes:3110 (3.0 KiB)
          Interrupt:40

2015-11-12 19:00:29.431357: INFO: TAP: Eth MAC address : b0:d5:cc:f1:a9:60
2015-11-12 19:00:29.431626: INFO: ETH: Eth MAC address : b0:d5:cc:f1:a9:60
2015-11-12 19:00:29.431730: INFO: LLSEC: Using 802.15.4 framer
2015-11-12 19:00:29.431926: INFO: LLSEC: Using 'nullsec' llsec driver
2015-11-12 19:00:29.433472: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e07:c5c3
2015-11-12 19:00:29.433901: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:e07:c5c3
2015-11-12 19:00:29.434117: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2015-11-12 19:00:29.434253: INFO: 6LBR: RA Daemon enabled
2015-11-12 19:00:29.434319: INFO: 6LBR: Checking addresses duplication
2015-11-12 19:00:41.124181: INFO: 6LBR: Configured as DODAG Root fd00::212:4b00:e07:c5c3
2015-11-12 19:00:41.125096: INFO: 6LBR: Starting as RPL ROUTER
2015-11-12 19:00:41.137223: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2015-11-12 19:00:41.141361: INFO: HTTP: Starting webserver on port 80
2015-11-12 19:00:41.142965: INFO: NODECFG: Node Config init
2015-11-12 19:00:41.143341: INFO: NODECFG: No node_config.conf file specified
2015-11-12 19:00:41.143769: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2015-11-12 19:00:41.147071: INFO: UDPS: UDP server started
CoAP push started
CoAP server started
2015-11-12 19:00:41.148134: INFO: DNS: DNS proxy started
2015-11-12 19:00:41.148395: INFO: PLUGIN: Initialising dummy
2015-11-12 19:00:41.150838: INFO: DUMMY: Dummy init
2015-11-12 19:00:41.151143: INFO: PLUGIN: Initialising lwm2m-client
2015-11-12 19:00:41.151342: INFO: LWM2M: LWM2M Client init
2015-11-12 19:00:41.151825: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2015-11-12 19:00:41.152180: INFO: 6LBR: CETIC 6LBR Started
RD client started
`
As you can see, nvm.conf file is not created. However, 6lbr starts properly and I can get the ip address in 6lbr.ip as bbbb::100. When I am using command ping bbbb::100, then i am getting error.

Can anyone please help me out or guide me the proper solution.

jacekrz, 2017-09-05:
Hi
two things:
1. `MODE=ROUTER - first char is '`' so remove it
2. I don't know Beabgle bone but file interfaces consists strange line:
      iface usb0 inet static
Could you show ifconfig?
JacekRz

jacekrz, 2017-09-05:
Hi
Ad 1. the first char in line MODE=ROUTER is `. Remove it.

closed 2017-11-09 by laurentderu