cetic/6lbr #323: Destination and source address swap in uip6.c

michelesponchiado, 2017-09-06:
I was trying to browse the 6lbr internal website across the 6lowPAN network:
`PC1↔6lbr of PC1↔contiki↔WSN↔contiki 2↔6lbr of PC2↔PC2↔le me browsing`
and the weird thing happening was that I was able to browse using the WSN address `fd00::x` but not using the IPV6 address` bbbb::100/cccc::100`; furthermore, the css style sheet was not loaded and the loading was very slow.

Checking the log I found out that the TCP requests sent from` cccc::100` to `bbbb::100` were replied to `cccc::100` from `fd00::1` and not from `bbbb::100`.
Checking the uip6.c near the `tcp_send` label, I saw this code:
```
uip_ipaddr_copy(&UIP_IP_BUF->destipaddr, &uip_connr->ripaddr);
uip_ds6_select_src(&UIP_IP_BUF->srcipaddr, &UIP_IP_BUF->destipaddr);
```
It seems that the `uip_ipaddr_copy(a,b)`just copies into _a_ the value of _b_.
So after the `uip_ipaddr_copy` is called, _UIP_IP_BUF->destipaddr_ value is set to _uip_connr->ripaddr_.
The second call tries to get the srcipaddress checking the best match of _UIP_IP_BUF->destipaddr_ that has just been set to the address of the sender of the TCP/IP message so the matching routine was returing the wrong result (IMHO)...
I just swapped the calls:
```
uip_ds6_select_src(&UIP_IP_BUF->srcipaddr, &UIP_IP_BUF->destipaddr);
uip_ipaddr_copy(&UIP_IP_BUF->destipaddr, &uip_connr->ripaddr);
```
and this way I can browse using the IPV6 addresses.
About the css not loading, I suppose this was a bug related to the incorrect sizing of the number of fragment buffer in sicslowpan.c but am I still checking.

michelesponchiado, 2017-09-15:
An improvement, to avoid problems if a retransmit is required, please use the following snippet to set destination and source IP addresses:

```
  if (memcmp((const void *)&UIP_IP_BUF->destipaddr, (const void *)&uip_connr->ripaddr, sizeof(UIP_IP_BUF->destipaddr)))
  {
	uip_ds6_select_src(&UIP_IP_BUF->srcipaddr, &UIP_IP_BUF->destipaddr);
  }
  uip_ipaddr_copy(&UIP_IP_BUF->destipaddr, &uip_udp_conn->ripaddr);

```

laurentderu, 2017-09-29:
There was indeed a problem with the source address selection, but the root cause is not the sequence of uip_ipaddr_copy() and uip_ds6_select_src(). Actually, the problem is due that uip_ds6_select_src should select the address using also the interface that will send the packet (as specified by RFC-6724 Rule 5), but in Contiki/6LBR the interface selection is done afterward, hence the source address selection is wrong.

I modified uip_ds6_select_src() to check if we have multiple interfaces (as the case of Router mode) and if so, perform a crude but effective interface checking beforehand. This also fix the problem for UDP and ICMP communication.

It's available in dev-1.4.x branch with commit https://github.com/cetic/6lbr/commit/2984106e226b81b89f6f01042b1105804a7d7b41 and develop branch with commit https://github.com/cetic/6lbr/commit/8a508c3b575408f67d0d0662d08db127dd4cd4e8

michelesponchiado, 2017-09-29:
Many thx Laurent,
you are doing an amazing job!

closed 2017-11-09 by laurentderu