cetic/6lbr #326: how to configure and build 6lbr with openwrt?

jchari, 2017-09-13:
Hi,
I was tried to configure 6lbr in openwrt as mentioned in wiki[0] page. i have observe 6lbr feeds successfully installed. but i am not able to build the 6lbr_ _ _.ipk. It is default,  taking as  'Y' to include it into the generated firmware. i am following below commands to compile package. please guide me, if i made any wrong. 

Then append the following line into openwrt feeds:
          src-git sixlbr https://github.com/cetic/6lbr-openwrt.git
         
         $ ./scripts/feeds update -a
         $ ./scripts/feeds install -a

         $ make menuconfig

         $ make package/6lbr/compile
         $ make package/6lbr/install
finally, 
         $ make V=s -j4
image was successfully built. but i didn't find 6lbr package installed.

[0]https://github.com/cetic/6lbr/wiki/OpenWRT-Installation

jchari, 2017-09-19:
Hi,

i am not able to find the 6lbr package in menuconfig and even not able to compile 6lbr package.

my issue has been solved with small correction in Makefile.
please find commit in below link,

https://github.com/cetic/6lbr-openwrt/pull/5/commits/9cd581cb63a623305e4da93afba66e0e52161235

closed 2017-09-19 by jchari

jchari, 2017-09-26:
Hi,

please merge this for @develop branch also. i found that, it is also have the same issue.

laurentderu, 2017-09-28:
Merged also in develop and dev-1.4.x