cetic/6lbr #332: tinydtls.h and dtls_config.h cannot be found in running 6lbr-demo

Mostafa-P, 2017-09-29:
when I want to run 6lbr-demo Fatal error appears that tinydtls.h does not exist. I find tinydtls.h.in and renamed it to tinydtls.h but another error appears that dtls_config.h can not find.
please help me to solve this problem.

laurentderu, 2017-09-29:
You must run the reconf.sh script in apps/tinydtls (or run autoconf and configure) to create the configuration files.

closed 2017-09-29 by laurentderu

Mostafa-P, 2017-09-30:
Thanks for your reply.
I only want to run DTLS (or SHA2 in any way) with wismote. because of this I try to use 6lbr-demo with 
# Applications

WITH_SHELL?=0
WITH_UDPCLIENT?=1
WITH_WEBSERVER?=0
WITH_COAPSERVER?=1
WITH_TINYDTLS?=1
WITH_DTLSECHO?=0
WITH_DTLS_COAP?=1
WITH_LLSEC?=0
WITH_TCP?=0
WITH_COAP_PROFILE?=IPSO_APP_FW
but I saw above error. may I know that my way is true ? and is it the simpelest way to implement DTLS(or SHA2) with wismote?

if it is true: I can not run this command.
I read  these links but can not help me:
https://github.com/cetic/6lbr/issues/139
https://github.com/dogecoin/dogecoin/issues/529
https://github.com/cetic/6lbr/wiki/Example-:-Dtls-Coap-Server
my sub-files  are update but I can not configure it.  this error appears:
user@instant-contiki:~$ cd 6lbr/apps/tinydtls
user@instant-contiki:~/6lbr/apps/tinydtls$ sh reconf.sh
: not found2: reconf.sh: 
: not found3: reconf.sh: autoreconf
: not found4: reconf.sh: 
reconf.sh: 5: reconf.sh: ./configure: not found
 
because of this I decide to use tinydtls file of contiki-dtls-tls and replaced it with 6lbr module but it is not working.
I need your reply urgently.
Thanks
thanks