cetic/6lbr #335: Not able to ping mesh nodes (cc2650) from Edge Router (BBB).

santhosh-chowdary, 2017-10-03:
I am trying to setup the 6Lowpan network with Beaglebone Black (Linux Debian) as edge router, CC2531 USB Dongle as SLIP Radio and two cc2650 flashed with cc26xx-web-demo as mesh nodes.

I am able to to see web server of BBB on browser. I am also able ping from my PC (cmd prompt) to BBB.
But I am not able to ping the mesh nodes from BBB. Web server (bbbb::100) also shows the two mesh nodes in sensor list.

Any of your suggestions would be appreciated and helpful.

Here are required configurations and images:

# 6lbr.conf:

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200
LOG_LEVEL=3 

# interafces:

/etc/network/interfaces:
#This file describes the network interfaces available on your system
#and how to activate them. For more information, see interfaces(5).

#The loopback network interface
auto lo
iface lo inet loopback

#The primary network interface
#auto eth0
#iface eth0 inet dhcp
#Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

##connman: ethX static config
#connmanctl services
#Using the appropriate ethernet service, tell connman to setup a static IP address for that service:
#sudo connmanctl config <service> --ipv4 manual <ip_addr> <netmask> <gateway> --nameservers <dns_server>

##connman: WiFi
#
#connmanctl
#connmanctl> tether wifi off
#connmanctl> enable wifi
#connmanctl> scan wifi
#connmanctl> services
#connmanctl> agent on
#connmanctl> connect wifi_*_managed_psk
#connmanctl> quit

##Ethernet/RNDIS gadget (g_ether)
#Used by: /opt/scripts/boot/autoconfigure_usb0.sh
iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.252
    network 192.168.7.0
    gateway 192.168.7.1

# nvm data:

root@beaglebone:~# /usr/lib/6lbr/bin/nvm_tool --print /etc/6lbr/nvm.dat
Reading nvm file '/etc/6lbr/nvm.dat'

Channel : 26
PAN ID : abcd

WSN network prefix : fd00::
WSN network prefix length : 64
WSN IP address : fd00::100
WSN accept RA : False
WSN IP address autoconf : True
WSN 6LoWPAN context 0 : fd00::
DNS server : ::

Eth network prefix : bbbb::
Eth network prefix length : 64
Eth IP address : bbbb::100
Eth default router : ::
Eth IP address autoconf : False

RA daemon : True
RA router lifetime : 0
RA maximum interval : 600
RA minimum interval : 200
RA minimum delay : 3
RA PIO enabled : True
RA prefix valid lifetime : 86400
RA prefix preferred lifetime : 14400
RA RIO enabled : True
RA RIO lifetime : 1800

RPL instance ID : 30
RPL DODAG ID manual : False
RPL DODAG ID : ::
RPL DODAG ID global : True
RPL Preference : 0
RPL version ID :  : 16
RPL DIO interval doubling : 8
RPL DIO minimum interval : 12
RPL DIO redundancy : 10
RPL default lifetime : 30
RPL maximum rank increase : 1792
RPL minimum rank increment : 256
RPL lifetime unit : 256

RPL Non Storing : False
RPL DAO ACK : True
RPL DAO ACK local repair : False
DIO triggers route refresh : False
MAC layer : 1
Security layer : 0
Security level : 6
Network key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
Noncoresec anti-replay enabled (Legacy) : False
Noncoresec anti-replay workaround (Legacy) : False
Filter unknown nodes : False
Disable NUD : True

NAT 64 Enabled : False
NAT 64 DHCP Enabled : False
NAT 64 static port mapping Enabled : False
NAT 64 RFC-6052 prefix : False
NAT 64 address : 172.16.0.2
NAT 64 netmask : 255.255.255.0
NAT 64 gateway : 172.16.0.1

Local address rewrite : True
Smart Multi BR : False
Webserver configuration page disabled : False
Webserver port : 80

# 6LBR Service start:

root@beaglebone:~# sudo service 6lbr start
root@beaglebone:~# sudo service 6lbr status
● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr; generated; vendor preset: enabled)
   Active: active (running) since Tue 2017-10-03 00:48:24 UTC; 4s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2129 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/6lbr.service
           ├─2137 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ├─2141 /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200
           ├─2142 /bin/sh /usr/lib/6lbr/6lbr-watchdog 2141 60 3 /var/log/6lbr.timestamp
           └─2187 sleep 1

Oct 03 00:48:24 beaglebone systemd[1]: Starting LSB: 6LoWPAN Border Router...
Oct 03 00:48:24 beaglebone 6lbr[2129]: Starting 6LoWPAN Border Router:.
Oct 03 00:48:24 beaglebone systemd[1]: Started LSB: 6LoWPAN Border Router.
root@beaglebone:~#

[image]

[image]

[image]

[image]

[image]

santhosh-chowdary, 2017-10-03:
Adding data of ifconfig and netstat -rn to above:

[image]

[image]

chenek, 2017-10-09:
For this you will need to configure your Ethernet interface for "bridge mode" as described in https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

ProfEng, 2017-10-11:
I have the same problem and couldn't figure it out. Can someone please give more information about it? In router mode, I can see nodes in webserver but it is not possible to ping them.

What can be wrong? How can I ping nodes from host(Raspberry Pi) which 6LBR runs.

Thanks in advance!

Aykut.

laurentderu, 2017-10-11:
The most probable cause is that you don't have a route towards the WSN network configured, check using the next command to see if there is a route towards fd00::/64 via bbbb::100

   ip -6 route

If not you need to manually create it or allow automatic route creation by setting ITF_AUTOCONF=1 in your 6lbr.conf file (see https://github.com/cetic/6lbr/wiki/6LBR-Connection)

santhosh-chowdary, 2017-10-15:
I have started 6lbr service (setup-BBB-edge router+CC2531- slip radio+CC2650 mesh node).
I have added a line "ITF_AUTOCONF=1" in my 6lbr.conf file and restarted 6lbr service.
And then executed command "ip -6 route" as you said. and  below are the results:

root@beaglebone:~# sudo service 6lbr status
● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr; generated; vendor preset: enabled)
   Active: active (running) since Sun 2017-10-15 01:59:11 UTC; 2min 21s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2616 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/6lbr.service
           ├─2624 /bin/sh /usr/lib/6lbr/6lbr /etc/6lbr/6lbr.conf
           ├─2628 /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyACM0 -r -t eth0 -B 115200
           ├─2629 /bin/sh /usr/lib/6lbr/6lbr-watchdog 2628 60 3 /var/log/6lbr.timestamp
           └─3355 sleep 1

Oct 15 01:59:10 beaglebone systemd[1]: Starting LSB: 6LoWPAN Border Router...
Oct 15 01:59:11 beaglebone 6lbr[2616]: Starting 6LoWPAN Border Router:.
Oct 15 01:59:11 beaglebone systemd[1]: Started LSB: 6LoWPAN Border Router.

 root@beaglebone:~#
 root@beaglebone:~# ip -6 route
 2003:e5:3bed:d101::/64 dev eth0 proto kernel metric 256  expires 13356sec mtu 1492 pref medium
 fe80::/64 dev usb0 proto kernel metric 256  pref medium
 fe80::/64 dev eth0 proto kernel metric 256  mtu 1492 pref medium
 root@beaglebone:~#

santhosh-chowdary, 2017-10-15:
Also  I tried adding the route manually by executing below commands in BBB ssh:
root@beaglebone:~# sudo ip -6 addr add bbbb::101/64 dev eth0
root@beaglebone:~# route -A inet6 add fd00::/64 gw bbbb::100

and then restarted 6lbr service. Now "ip -6 route" shows as below:
root@beaglebone:~# ip -6 route
2003:e5:3bed:d101::/64 dev eth0 proto kernel metric 256  expires 14195sec mtu 1492 pref medium
bbbb::/64 dev eth0 proto kernel metric 256  pref medium
fd00::/64 via bbbb::100 dev eth0 metric 1  pref medium
fe80::/64 dev usb0 proto kernel metric 256  pref medium
fe80::/64 dev eth0 proto kernel metric 256  mtu 1492 pref medium
root@beaglebone:~#

and then I pinged the node with ipv6 address -  fd00::212:4b00:6e2:9f89 and below are the results:
root@beaglebone:~# ping6 fd00::212:4b00:6e2:9f89
PING fd00::212:4b00:6e2:9f89(fd00::212:4b00:6e2:9f89) 56 data bytes
From bbbb::101 icmp_seq=1 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=2 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=3 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=4 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=5 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=6 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=7 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=8 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=9 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=10 Destination unreachable: Address unreachable
From bbbb::101 icmp_seq=11 Destination unreachable: Address unreachable
^C
--- fd00::212:4b00:6e2:9f89 ping statistics ---
17 packets transmitted, 0 received, +15 errors, 100% packet loss, time 16088ms

Or can I ping mesh nodes by setting up in smart bridge mode ?

santhosh-chowdary, 2017-10-15:
@chenek
I have configured my Ethernet interface for bridge mode as you said, but i am not able browse web server page now.

My 6lbr.conf file is :
#MODE=ROUTER
MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=1
DEV_ETH=eth0
DEV_BRIDGE=br0
DEV_TAP=tap0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

LOG_LEVEL=3

And my /etc/network/interfaces is :

# The loopback network interface
auto lo
iface lo inet loopback

# Ethernet/RNDIS gadget (g_ether)
# Used by: /opt/scripts/boot/autoconfigure_usb0.sh
iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.252
    network 192.168.7.0
    gateway 192.168.7.1

iface eth0 inet static
address 0.0.0.0

auto br0
iface br0 inet dhcp
    bridge_ports eth0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`

Also I am not able to ping bbbb::100 from my PC now.

santhosh-chowdary, 2017-10-16:
@ProfEng , If you have any progress in above , please let me know.

santhosh-chowdary, 2017-10-16:
@laurentderu , @chenek  : Waiting for your suggestions.

laurentderu, 2017-10-17:
@santhosh-chowdary if you are using a BBB, the bridge interfacing (eth0 - br0 - tap0) does not work because of driver limitation in the Linux kernel. You either have to use Raw Ethernet mode (but then you can't access 6LBR from the BBB), or Route interfacing, see https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration

Note that even if the name are similar, bridge interfacing has nothing to do with Smartbridge mode, same for Route interfacing and Router mode !

santhosh-chowdary, 2017-10-17:
@laurentderu , 
1) You said I can't access 6LBR from the BBB in Raw Ethernet mode. Do you mean that I can't access 6lbr webserver or I can't ssh to BBB and access 6lbr to start 6lbr service?

2) So, If I want to use Route interfacing which is not same as Router mode ,w which one should I use. Do you mean I have to use "Routing mode" given in link https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration.

3) http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples explains to use BBB only.

4) If I buy Raspberry Pi , will it easier than BBB to implement edge router and 6Lowpan Network setup ? If Raspberry Pi is easier, which mode is better? 

My aim is to send sensor data from CC2650 mesh node to  edge router over 802.15.4 and then to internet. Any suggestions on this to accomplish?

ProfEng, 2017-10-18:
@santhosh-chowdary I made it happen with raspberry pi (2 and zero) easily in router mode. If you buy one, I can let you know how to do it. Btw I am using mqtt in nodes and raspberry to communicate each other. If you want another setup, I can explain you how to ping nodes (so: reaching nodes) and you can go on.

santhosh-chowdary, 2017-10-18:
@ProfEng, I would definitely appreciate that. Can you tell me. what is your setup? I will buy raspberry. Kindly let me know the steps. "santhu.manik@gmail.com" is my gmail ID please send me the steps.
Also Do I have to but Raspberry 2.0 or latest one (raspberry 3.0)?

chenek, 2017-10-23:
Do you run "sudo route -A inet6 add fd00::/64 gw bbbb::100" on your BBB?

santhosh-chowdary, 2017-10-23:
@chenek , Yes I did that but still no use. Right now I am trying to build slip radio with nullrdc setting for CC2531 USB Dongle as per https://github.com/cetic/6lbr/issues/234

Need to check.

chenek, 2017-10-23:
Try to refer to the steps in http://sunmaysky.blogspot.tw/2017/10/how-to-configure-6lbr-to-make-it-can-do.html and test again.

ProfEng, 2017-10-23:
Install Raspbian Image (Latest Version)
Install 6LBR from develop branch as detailed in 6LBR wiki (1.5.x develop branch used)
In /etc/dhcpcd.conf add line:  
denyinterfaces eth0 tap0 wlan0 (Raspberry Pi2)
denyinterfaces tap0 (Raspberry Pi Zero Wireless)
6LBR.conf must be like this: 

MODE=ROUTER
RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyUSB0
BAUDRATE=115200
LOG_LEVEL=3 

Add ip address and routes manually with the commands:
sudo ip -6 addr add bbbb::101/64 dev eth0

route -A inet6 add fd00::/64 gw bbbb::100

From now on, 6LBR must be worked and the nodes can be pinged from Raspberry.

santhosh-chowdary, 2017-10-24:
@ProfEng ,
How about Slip radio and node? 
1) Are you using CC2531 usb dongle (cc2531-slip-radio_contikimac.hex) as slip radio?
   channel - 25 and PAN ID - 0xABCD.
2) Are you using srf06+CC2650EM (cc26xx-web-demo.bin) as node?
   channel - 25 and PAN ID - 0xABCD.

If possible can you share me cc2531-slip-radio_contikimac.hex and cc26xx-web-demo.bin files you have used.

Also Can I know the initial Raspberry PI connections? 
I need to connect the 
1) power cable to raspberry pi and 
2) RJ45 LAN cable from router to Ethernet port of raspberry pi and 
3) HDMI cable from raspberry pi to PC 
right?

santhosh-chowdary, 2017-10-24:
@chenek 

I have followed all steps carefully for BBB, from your above blog as you said. 
But in Smart bridge mode , I can't even browse web server page http://bbbb::100. 
Also /var/log/6lbr.ip has fd00::212:4b00:768:d35b address instead of bbbb::100  address.

ProfEng, 2017-10-24:
@santhosh-chowdary 

I didn't make any change at /etc/network/interfaces and I am using at router mode. 

1 - I am using Openmote (cc2538) as slip-radio and I compiled it myself from examples/ipv6/border-router

2 - My node is Openmote,too. But I am using mqtt example and it is not important. It must be pingable anyway. 

And finally, yes. your connections are right.

santhosh-chowdary, 2017-10-27:
@ProfEng 

You have mentioned that, you are using cc2538 as slip radio and compiled it yourself from examples/ipv6/border-router. Is that border-router or examples/ipv6/slip-radio ??

I am not able to compile examples/ipv6/slip-radio for cc2650 as target (am using contiki 3).

santhosh-chowdary, 2017-10-27:
@ProfEng 
I have bought latest Raspberry 3 B(wireless) and tried as you said.

Connections: 
1) power cable to raspberri pi 3.
2) RJ45 LAN cable from router to raspberry pi Ethernet port.
3) CC2531 slip radio to raspberri pi USB.
4) I took SSH session of RPI3 from windows PC connected to internet via Wifi to the same network as RPI3.
 
Installed Raspbian Image (Latest Version)
Installed 6LBR from develop branch (1.5.x develop branch used)

In /etc/dhcpcd.conf added line:
denyinterfaces tap0  (since Raspberry Pi 3 )

6LBR.conf :

MODE=ROUTER
RAW_ETH=0
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0   ## as I am using CC2531 as slip radio 2.4 GHZ
BAUDRATE=115200
LOG_LEVEL=3

And then i started 6lbr services
But i didnt get 6lbr web page at all on browser. I could not ping bbbb::100 from windows PC.

Then I only changed 6lbr.conf as below:

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0   ## as I am using CC2531 as slip radio 2.4 GHZ
BAUDRATE=115200
LOG_LEVEL=3

and remaining all settings  as same. 
Then I restarted 6lbr services  , now i could browse 6lbr web page from windows PC.
But again I could not ping the mesh nodes(cc26xx-web-demo.bin)

Strange behavior is, after adding below routes:  

sudo ip -6 addr add bbbb::101/64 dev eth0
route -A inet6 add fd00::/64 gw bbbb::100

pi@raspberrypi:~ $ ip -6 route
2003:e5:3bd6:ad01::/64 dev eth0 proto kernel metric 202  mtu 1492 pref medium
2003:e5:3bd6:ad01::/64 dev wlan0 proto kernel metric 303  mtu 1492 pref medium
bbbb::/64 dev eth0 proto kernel metric 256  pref medium
bbbb::/64 dev wlan0 proto kernel metric 303  mtu 1500 pref medium
fd00::/64 via bbbb::100 dev eth0 metric 1  pref medium
fe80::/64 dev wlan0 proto kernel metric 256  pref medium
fe80::/64 dev eth0 proto kernel metric 256  pref medium
default via fe80::1 dev eth0 metric 202  mtu 1492 pref medium
default via fe80::1 dev wlan0 metric 303  mtu 1492 pref medium

I could not ping bbbb::100 from raspberrypi ssh session, but I could ping from my windows PC.

pi@raspberrypi:~ $ ping6 fd00::212:4b00:6e2:3b0b
PING fd00::212:4b00:6e2:3b0b(fd00::212:4b00:6e2:3b0b) 56 data bytes
From bbbb::101 icmp_seq=1 Destination unreachable: Address unreachable
^C
--- fd00::212:4b00:6e2:3b0b ping statistics ---
11 packets transmitted, 0 received, +1 errors, 100% packet loss, time 10395ms

And I have got below error in /var/log/6lbr.log:
2017-10-26 21:45:42.180826: ERROR: PFE: wireless_output: sending to self
2017-10-26 21:45:42.181981: ERROR: PFE: wireless_output: sending to self

Slip radio: 
CC2531 USB Dongle as  slip radio flashed with  "cc2531-slip-radio_contikimac.zip".

Mesh node:
srf06+ccc2650EM flashed with cc26xx-web-demo.bin (channel 25, nullrdc, UIP_CONF_ROUTER 1, UIP_CONF_ND6_SEND_NA 1).

Below are 6lbr web page images for referrence:

[image]

[image]

[image]

Still no luck , it seems like I am facing all the possible errors, Any help

santhosh-chowdary, 2017-10-29:
@laurentderu , @ProfEng . @chenek 
Can anybody please explain me in full detail about the addresses displayed under system-tab(like addresses under WSN and Ethernet headings) and status-tab(Addresses, IP64, Neighbors, Routes heading) of 6lbr web page. I want to know about each address and what hardware device do they exactly mean? 

Here are my system tab and status tab:
[image]

[image]

santhosh-chowdary, 2017-11-04:
@ProfEng , @laurentderu , @chenek 
Kindly reply. I am waiting for your suggestions.

laurentderu, 2017-11-09:
In Raw Ethernet mode, you can not reach the 6LBR from the host. You have to use Bridging interface mode or Routing interface mode (note these have nothing to do with Router or Smartbridge mode, I know the name could be confusing)

On the RPi, to be able to reach the BR using bridging interface mode and Router mode, the br0 interface (and not eth0) must have a bbbb::/64 address and you must have a route towards fd00::/64

Theoretically, using ITF_AUTOCONF=1 ensure that, but you need to check using the commands 

    ip addr br0
    ip -6 route

Note that in Smartbridge mode, the same subnet is used on both the WSN segment and Ethernet segment, so the BR does not hava a bbbb::54 address but a fd00::/64 based on its mac address.

ProfEng, 2017-11-09:
Hi @santhosh-chowdary 

Can you share please:

 What is your Raspberry Pi number (2-3-zero) ? 

What did you add to the dhcpcd.conf (denyinterfaces ?)?

And a ss from ifconfig from raspberry.

santhosh-chowdary, 2017-11-12:
@ProfEng 
I am using Raspberry PI 3, model B.
Raspbian version in below screenshot:
[image]

I have added in dhcpcd.conf:
denyinterfaces tap0

# 6lbr.conf:

MODE=ROUTER
RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
RAW_ETH_FCS=1
DEV_RADIO=/dev/ttyACM0
BAUDRATE=115200
LOG_LEVEL=3

Result : I can browse bbbb::100 from browser, But I can't ping sensor nodes.

screenshot ifconfig from raspberry:
[image]

I have tried the below config in 6lbr.conf

mode:router
bridge:0
raw_eth:0

in dhcpcd.conf
denyinterfaces tap0

Result : I can't browse bbbb::100 from browser.

lucasscola, 2017-11-28:
There was a problem with the Contiki version used in the TI demo and the one used in 6lbr. Check both Context 0 prefix, they're probably different. Let me check the demo that I ran once in order to see the details, but I remember having your same problem.

santhosh-chowdary, 2017-11-29:
@lucasscola ,
Do you mean that version of 6lbr and contiki version used to build cc26xx-web demo have the problem?
What do you mean by Context 0 prefix ?
I am using : 
6lbr Version : 1.5.x (Contiki-contiki-base-develop-20170120-2419-g1606dd3eb).
and 
cc26xx-web-demo(TI) version - Instant Contiki 3.0
Is there a problem with these above versions?

lucasscola, 2017-11-29:
In the version I used, by default context-based header compression (HC06) was used in 6LoWPAN (where the WSN network prefix is omitted in the packets and that info is sent as "context", with less bits). The problem is that, for this compression to work the node and the gateway must have the same context in order to decompress the IPv6 header.
Take a look in your 6LBR implementation:
> WSN network prefix : fd00::

Old versions of Contiki use aaaa:: as network prefix so check which version are you using to compile the nodes.
https://github.com/contiki-os/contiki/commit/dea04c67d79e2bec92382340be6df70de90d3210#diff-5866ac53ae79a842d83563f356597f6b

If you are using a newer version, this migth not be the issue, and its more likely a networking problem in the 6lbr implementation.

laurentderu, 2017-12-14:
@lucasscola is right, Contiki-3.0 still uses prefix aaaa::/64, so you must either recompile your nodes using a later version of Contiki or change the prefix compiled; or change the prefix in Configuration page of 6LBR.

gani-koduri, 2018-03-25:
"sudo route -A inet6 add fd00::/64 gw bbbb::100" 

==> this solved my issue. able to reach WSN IPs after adding this.

chenek, 2018-03-26:
@gani-koduri I had tried "sudo route -A inet6 add fd00::/64 gw bbbb::100" but always get "SIOCADDRT: Invalid argument" error.

rishabhj14, 2018-05-19:
Hi everyone,
I and my team are working on the same project above.
I have made the configuration for 6lbr.conf file similar to those above. 
I can ping my node from my windows PC but unable to see the parent node.
Also When I flash my node with web-demo.bin file it does not show node on 6lbr but if I flash it with demo.bin file it shows the node on 6lbr.
Can Anyone one help me plzzzzzz??

chenek, 2018-05-19:
@rishabhj14 Try to refer https://sunmaysky.blogspot.tw/2017/10/how-to-configure-6lbr-to-make-it-can-do.html?m=1