cetic/6lbr #341: 6Lowpan network Implementation

santhosh-chowdary, 2017-10-17:
Untill Now , Has Anyone succeeded with implementation of 6Lowpan network using 6lbr ?
whether the Setup is with BBB/ Raspberry (edge router) + CC2531/CC2650/CC1330 (slip) + CC2650EM/CC2650stk (mesh nodes).?

Please share us the steps and content of the files like :
/etc/network/interfaces
/etc/6lbr/6lbr.conf
any important tweeks , settings, changes in cc26xx-web-demo?  

I am trying the same with Router mode , I am able to get the 6lbr web page but I am not able to ping to mesh node from my BBB.

If I try in smart bridge mode as in link ( https://github.com/cetic/6lbr/wiki/6LBR-Interface-Configuration) , I wont even get the 6lbr web page.

Kindly help me in this implementation. i have been trying all the ways but not able to succeed till now.

chenek, 2017-10-23:
Try to refer to http://sunmaysky.blogspot.tw/2017/10/how-to-configure-6lbr-to-make-it-can-do.html

chenek, 2017-11-08:
@santhosh-chowdary if you already make it, please close the issue.

closed 2017-11-09 by laurentderu