cetic/6lbr #350: alignment trap error  use route mode compile in arm 926e cpu with gcc 6.2

junwushi, 2017-12-08:
root@gateway:~# [  564.320000] Alignment trap: cetic_6lbr_rout (3470) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  575.940000] Alignment trap: cetic_6lbr_rout (3529) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  584.250000] Alignment trap: cetic_6lbr_rout (3603) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  592.670000] Alignment trap: cetic_6lbr_rout (3662) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  600.300000] Alignment trap: cetic_6lbr_rout (3721) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  611.860000] Alignment trap: cetic_6lbr_rout (3780) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  619.570000] Alignment trap: cetic_6lbr_rout (3854) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  627.090000] Alignment trap: cetic_6lbr_rout (3913) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  634.610000] Alignment trap: cetic_6lbr_rout (3967) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  642.430000] Alignment trap: cetic_6lbr_rout (4021) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  650.550000] Alignment trap: cetic_6lbr_rout (4080) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  658.580000] Alignment trap: cetic_6lbr_rout (4139) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x801
[  666.400000] Alignment trap: cetic_6lbr_rout (4198) PC=0x0005b610 Instr=0xe58c5004 Address=0x001eab46 FSR 0x8

laurentderu, 2017-12-14:
Could you provide more context related to the error ?
It would help a lot if you could connect a debugger to the 6lbr process and get the backtrace of the error.

junwushi, 2017-12-18:
below that i  fixed code  with alignment trap issue in /6lbr/core/net/ipv6/uip-nd6.c.
/*
 * Copyright (C) 1995, 1996, 1997, and 1998 WIDE Project.
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the project nor the names of its contributors
 *    may be used to endorse or promote products derived from this software
 *    without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE PROJECT AND CONTRIBUTORS ``AS IS'' AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED.  IN NO EVENT SHALL THE PROJECT OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS
 * OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY
 * OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 */
/*
 * Copyright (c) 2006, Swedish Institute of Computer Science.
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *   notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *   notice, this list of conditions and the following disclaimer in the
 *   documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the Institute nor the names of its contributors
 *   may be used to endorse or promote products derived from this software
 *   without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE INSTITUTE AND CONTRIBUTORS ``AS IS'' AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED.  IN NO EVENT SHALL THE INSTITUTE OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS
 * OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY
 * OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 *
 */

/**
 * \addtogroup uip6
 * @{
 */

/**
 * \file
 *    Neighbor discovery (RFC 4861)
 * \author Mathilde Durvy <mdurvy@cisco.com>
 * \author Julien Abeille <jabeille@cisco.com>
 */
[... 1210 lines cut ...]
uip_nd6_init()
{
#if UIP_ND6_SEND_NA
  /* Only handle NSs if we are prepared to send out NAs */
  uip_icmp6_register_input_handler(&ns_input_handler);
#endif

#if UIP_ND6_SEND_NS
  /*
   * Only handle NAs if we are prepared to send out NSs. */
  uip_icmp6_register_input_handler(&na_input_handler);
#endif

#if UIP_CONF_ROUTER && UIP_ND6_SEND_RA
  /* Only accept RS if we are a router and happy to send out RAs */
  uip_icmp6_register_input_handler(&rs_input_handler);
#endif

#if !UIP_CONF_ROUTER
  /* Only process RAs if we are not a router */
  uip_icmp6_register_input_handler(&ra_input_handler);
#endif
}
/*---------------------------------------------------------------------------*/
 /** @} */

junwushi, 2017-12-28:
i think that has alignment issue is below structure define:
#define UIP_IP_BUF ((struct uip_ip_hdr )&uip_buf[UIP_LLH_LEN]) /< Pointer to IP header */
#define UIP_ICMP_BUF ((struct uip_icmp_hdr *)&uip_buf[uip_l2_l3_hdr_len]) /< Pointer to ICMP header/
/**@{ Pointers to messages just after icmp header */
#define UIP_ND6_RS_BUF ((uip_nd6_rs *)&uip_buf[uip_l2_l3_icmp_hdr_len])
#define UIP_ND6_RA_BUF ((uip_nd6_ra )&uip_buf[uip_l2_l3_icmp_hdr_len])
#define UIP_ND6_NS_BUF ((uip_nd6_ns )&uip_buf[uip_l2_l3_icmp_hdr_len])
#define UIP_ND6_NA_BUF ((uip_nd6_na )&uip_buf[uip_l2_l3_icmp_hdr_len])
/ @} /
/ Pointer to ND option */
#define UIP_ND6_OPT_HDR_BUF ((uip_nd6_opt_hdr *)&uip_buf[uip_l2_l3_icmp_hdr_len + nd6_opt_offset])
#define UIP_ND6_OPT_PREFIX_BUF ((uip_nd6_opt_prefix_info *)&uip_buf[uip_l2_l3_icmp_hdr_len + nd6_opt_offset])
#define UIP_ND6_OPT_MTU_BUF ((uip_nd6_opt_mtu )&uip_buf[uip_l2_l3_icmp_hdr_len + nd6_opt_offset])
#define UIP_ND6_OPT_RDNSS_BUF ((uip_nd6_opt_dns )&uip_buf[uip_l2_l3_icmp_hdr_len + nd6_opt_offset])
/ @} */

laurentderu, 2018-01-05:
Thank you for the detailed feedback ! I made a more comprehensive fix, could you check if it works on your platform too ? It's available in this branch : https://github.com/cetic/6lbr/tree/fix-alignment

junwushi, 2018-01-19:
yes,update it ,but still remain it when start the 6lbr service.
starting webserver for 6lbr gateway
[   12.380000] usb 1-2: new high-speed USB device number 4 using nuc970-ehci
[   15.780000] Alignment trap: cetic_6lbr_smar (847) PC=0x00054f08 Instr=0xe78c000b Address=0x001e9022 FSR 0x801
[   16.830000] Alignment trap: cetic_6lbr_smar (847) PC=0x00054f08 Instr=0xe78c000b Address=0x001e9022 FSR 0x801
[   17.760000] usb 1-2: new high-speed USB device number 5 using nuc970-ehci
[   20.370000] Alignment trap: cetic_6lbr_smar (847) PC=0x00054f08 Instr=0xe78c000b Address=0x001e9022 FSR 0x801
[   23.160000] Alignment trap: cetic_6lbr_smar (847) PC=0x00054f08 Instr=0xe78c000b Address=0x001e9022 FSR 0x801

laurentderu, 2018-01-19:
And with your modifications you don't have them ? I may have missed one or two places.

junwushi, 2018-01-19:
yes,i am not found it,so not fixed it.because it is happened only when service starting,attach the progam to debug is difficult. after update the last ,there are many NR node appeared. like below:
[image]

junwushi, 2018-03-23:
Program received signal SIGBUS, Bus error.
0x00083168 in uip_nd6_ra_output (dest=0x0)
    at ./../../core/net/ipv6/uip-nd6.c:940
940			  UIP_ND6_OPT_ROUTE_BUF->rlifetime = uip_htonl(rtinfo->lifetime);