cetic/6lbr #361: Ubidots with srf06-cc13xx

BlueberryTech01, 2018-03-20:
Hi all, especially @g-oikonomou 

I am working with Contiki OS and with the srf06/cc13xx. Now my goal is to send data to the cloud. I saw a tutorial of yours on Hackster and a guide provided on GitHub: Https://github.com/g-oikonomou/contiki/tree/ubidots-demo/examples/ipv6/ubidots. I have now tried to perform all these steps and in the Makefile I am using Contiki Ipv6 with a border-router with wrapsix. When I connect the node to the terminal I see it tells me UBIDOTS CLIENT STATA_ERROR_NO_NET. What comes with this error? Thanks for the reply.

Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET
Ubidots client: STATE_ERROR_NO_NET

BlueberryTech01, 2018-03-27:
@g-oikonomou

g-oikonomou, 2018-03-28:
Hi Mario

I've not used that particular example for quite a while ago, pretty much since I developed it. I am not sure if the Ubidots API has changed since then, but the problem here seems to be caused by something much more fundamental. Your device cannot join a network, let alone send API requests to the ubidots cloud. Suggest you try a different example to make sure your device can join the network correctly before attempting more advanced things. The example is obviously based on top of a somewhat old code basis, you may benefit by trying it with a more recent one

BlueberryTech01, 2018-03-28:
Hi George

Could you suggest something to me? I need this kind of example. I would send data to the cloud Ubidots, but there are errors that prevent me

BlueberryTech01, 2018-03-28:
@alignan

BlueberryTech01, 2018-04-04:
I try to make tcpdump on my eth0 and I see this. Why not function?

[image]

BlueberryTech01, 2018-04-04:
@g-oikonomou

BlueberryTech01, 2018-04-04:
@alignan can you help me?

closed 2018-05-16 by BlueberryTech01