cetic/6lbr #362: Where do i see the mac address for a cc1310 node

chansdad, 2018-03-23:


chansdad, 2018-03-23:
so hav ethe 6lbr able to see the nodes , can see it assign ipv6 addresses , can ping6 them from raspberry pi etc , all good , now trying to add static coap port mapping and realized it is the mac id of the node that needs to be put in the node_config file , interestingly enough , i dont see to find a mac id for the cc1310 node 

Now i ams starting to scratch my head , will there be a mac id - frommy knowledge yes it should have one , then where is it , or am i wrong?

If i am wrong , what should i do to be able to see the mac id in the 6lbr webpage? 

Thanks

chenek, 2018-03-26:
It uses device MAC address and generated a unique IPv6 IP address. For example, if you have mac address  "00 12 4B 00 0A 57 F4 79", it will turn into IPv6 address "fd00::2**12:4b00:a57:f479**"