cetic/6lbr #366: 6lbr with contiki-ng build issue.

chenek, 2018-03-29:
I tried to build 6lbr with contiki-ng on Raspberry Pi 2 and the followings are my steps.

1. change directory to home folder and git clone --recursive https://github.com/cetic/6lbr.git
2. change directory to 6lbr folder and git checkout contiki-ng
3. change directory to home folder and git clone --recursive https://github.com/cetic/contiki-ng.git
4. change directory  to contiki-ng folder and git checkout sixlbr
5. change directory to 6lbr/examples/6lbr folder and run the following build command
          make WITH_CONTIKI=0 CONTIKI=../../../contiki-ng all

However, I get the following build error: What do I miss?

pi@raspberrypi:~/6lbr/examples/6lbr $ make WITH_CONTIKI=0 CONTIKI=../../../contiki-ng all
rm -rf bin
mkdir -p bin
make TARGET=native clean
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr'
rm -f *~ *core core *.srec \
*.lst *.map *.o \
*.cprg *.bin *.data contiki*.a *.firmware core-labels.S *.ihex *.ini \
rm -rf *.so obj_native/*.so *.native
rm -rf obj_native
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr'
make TARGET=native bin/cetic_6lbr_smart_bridge
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr'
make TARGET=native CETIC_6LBR_SMARTBRIDGE=1
make[2]: Entering directory '/home/pi/6lbr/examples/6lbr'
mkdir obj_native
  CC        ../../../contiki-ng/arch/platform/native/./platform.c
  CC        ../../../contiki-ng/arch/platform/native/./clock.c
  CC        ../../../contiki-ng/arch/platform/native/dev/xmem.c
  CC        ../../../contiki-ng/arch/platform/native/dev/leds-arch.c
  CC        ../../../contiki-ng/arch/platform/native/./cfs-posix.c
  CC        ../../../contiki-ng/arch/platform/native/./cfs-posix-dir.c
  CC        ../../../contiki-ng/arch/cpu/native/net/tun6-net.c
  CC        ../../../contiki-ng/arch/cpu/native/./rtimer-arch.c
  CC        ../../../contiki-ng/arch/cpu/native/./watchdog.c
  CC        ../../../contiki-ng/arch/cpu/native/dev/eeprom.c
  CC        ../../../contiki-ng/os/services/slip-cmd/cmd.c
  CC        ../../../contiki-ng/os/services/slip-cmd/packetutils.c
  CC        ../../../contiki-ng/os/net/mac/csma/csma.c
  CC        ../../../contiki-ng/os/net/mac/csma/csma-output.c
  CC        ../../../contiki-ng/os/net/mac/nullmac/nullmac.c
  CC        ../../../contiki-ng/os/services/ip64/ip64-dns64.c
  CC        ../../../contiki-ng/os/services/ip64/ip64-arp.c
  CC        ../../../contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c
In file included from ./project-conf.h:500:0,
                 from ../../../contiki-ng/arch/platform/native/./contiki-conf.h:38,
                 from ../../../contiki-ng/os/contiki.h:37,
                 from ../../../contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c:32:
./6lbr/conf-mapping.h:98:48: error: missing binary operator before token "("
 #define RPL_CONF_MOP (cetic_6lbr_config_rpl_mop())
                                                ^
../../../contiki-ng/os/net/routing/rpl-classic/rpl-conf.h:52:41: note: in expansion of macro ‘RPL_CONF_MOP’
 #define RPL_MOP_DEFAULT                 RPL_CONF_MOP
                                         ^
../../../contiki-ng/os/net/routing/rpl-classic/rpl-conf.h:85:28: note: in expansion of macro ‘RPL_MOP_DEFAULT’
 #if RPL_WITH_MULTICAST && (RPL_MOP_DEFAULT != RPL_MOP_STORING_MULTICAST)
                            ^
../../../contiki-ng/Makefile.include:305: recipe for target 'obj_native/ip64-ipv4-dhcp.o' failed
make[2]: *** [obj_native/ip64-ipv4-dhcp.o] Error 1
make[2]: Leaving directory '/home/pi/6lbr/examples/6lbr'
platform/native/Makefile.native:58: recipe for target 'bin/cetic_6lbr_smart_bridge' failed
make[1]: *** [bin/cetic_6lbr_smart_bridge] Error 2
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr'
platform/native/Makefile.native:93: recipe for target 'all' failed
make: *** [all] Error 2

chenek, 2018-04-13:
Git to latest source to test today and find issue is gone.

razeen123, 2018-04-24:
I also have the same issue

chenek, 2018-04-24:
Do you git to latest source code?

razeen123, 2018-04-24:
yes. latest code for 6lbr. i did this with following commands
sudo apt-get update
sudo apt-get install bridge-utils libncurses5-dev
git clone --recursive https://github.com/cetic/6lbr

razeen123, 2018-04-24:
My Raspbian version. working on a Raspberry Pi 3 B+

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

razeen123, 2018-04-25:
@chenek Thank you very much. it compiled.

chenek, 2018-04-25:
@razeen123 yes, you should create 6lbr.conf.

closed 2018-04-25 by chenek