cetic/6lbr #384: NullPointerException when running 6LBR+simulated WSN

zwgraham, 2018-10-25:
I'm trying to get up and running with 6lbr and the cooja simulator. I understand that this is an old wiki page, but I was referencing: https://github.com/cetic/6lbr/wiki/COOJA-Interface 

I followed the instructions on the page, with some additional troubleshooting on my own.

I'm running Ubuntu 18.04, java-8-oracle, and using the 6lbr develop branch.

### 

Steps to reproduce:
1. open a terminal window and navigate to 6lbr/examples/6lbr
2. run make all (all_native isn't a target)
3. run make tools
4. navigate into 6lbr/examples/6lbr/test
5. make cooja-small

note: Once the cooja window is open and the various panes are populated I get a pop-up message about compilation warnings, this has errors listed in it too. The contents of that popup are in [cooja-error.txt](https://github.com/cetic/6lbr/files/2512854/cooja-error.txt) and there are errors listed in terminal 1. In a nutshell, the errors are
```
javac 1.8.0_191
org/contikios/cooja/corecomm/Lib1.java:40: error: cannot find symbol
public class Lib1 extends CoreComm {
                          ^
  symbol: class CoreComm
1 error
```
6. close error window and open up a second terminal
7. in terminal 2 navigate to 6lbr/examples/6lbr/test
8. make sure a bridge exists, otherwise the next step will lead to errors `ip link add name br0 type bridge && ip link set br0 up`
9. in terminal 2 run `make launch-6lbr-router-1-cooja`
10. observe the log output until I see `INFO: BR-RDC: Fetching MAC address`
11. in cooja simulation window click start

This is where things fall apart.

Description of events after start button is pressed:
- `BR-RDC: Fetching MAC address` messages stops in terminal 2, as expected. 6lbr emits new logs
- the 'mote output' pane of cooja is populated with an entry for every mote
- I get the NullPointerError message in cooja
```
java.lang.NullPointerException
	at java.util.HashMap$HashIterator.nextNode(HashMap.java:1445)
	at java.util.HashMap$ValueIterator.next(HashMap.java:1471)
	at org.contikios.cooja.contikimote.ContikiMoteType.getCoreMemory(ContikiMoteType.java:814)
	at org.contikios.cooja.contikimote.ContikiMote.execute(ContikiMote.java:148)
	at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
	at org.contikios.cooja.Simulation.run(Simulation.java:290)
	at java.lang.Thread.run(Thread.java:748)
```
- as the error above is shown, 6lbr emits error logs as well (lines after `6lbr checking address duplication`) 

```
2018-10-24 18:21:20.379651: INFO: BR-RDC: Fetching MAC address
2018-10-24 18:21:20.380335: INFO: SLIP: Contiki-contiki-6lbr-1.5.0-2532-g2e5d6f8a3 started. Node id is set to 1.
Rime started with address 0.1.0.1.0.1.0.1
MAC 00:01:00:01:00:01:00:01 slipnet/nullmac/nullrdc, channel check rate 1000 Hz
Tentative link-local IPv6 address fe80:0000:0000:0000:0201:0001:0001:0001
Tentative global IPv6 address fd00:0000:0000:0000:0201:0001:0001:0001
Starting 'Slip radio process'
Slip Radio started...
2018-10-24 18:21:20.380426: INFO: BR-RDC: Got MAC 0 : 00:01:00:01:00:01:00:01
2018-10-24 18:21:21.779830: INFO: ETH: TAP Ethernet interface init
2018-10-24 18:21:21.779915: INFO: TAP: opened device tap0
2018-10-24 18:21:21.779952: INFO: TAP: Running 6lbr-ifup script '../package/usr/lib/6lbr/6lbr-ifup'
6lbr-ifup: Attach device tap0 to bridge br0
44: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 02:0a:0b:0c:0d:0e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::68b0:3eff:fe5e:f755/64 scope link 
       valid_lft forever preferred_lft forever
40: tap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br0 state UP group default qlen 1000
    link/ether 02:0a:0b:0c:0d:0e brd ff:ff:ff:ff:ff:ff
2018-10-24 18:21:21.832725: INFO: ETH: Eth MAC address : 00:01:00:01:00:01
[... 41 lines cut ...]
2018-10-24 18:21:32.835063: WARN: CONFIG: Can not open ../package/etc/6lbr/nvm.conf : No such file or directory
2018-10-24 18:21:32.835076: INFO: 6LBR: CETIC 6LBR Started
RD client started
2018-10-24 18:21:35.085623: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (9)
2018-10-24 18:21:35.285605: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (10)
2018-10-24 18:21:35.486079: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (11)
2018-10-24 18:21:35.685587: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (12)
2018-10-24 18:21:35.886133: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (13)
2018-10-24 18:21:36.086164: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (14)
2018-10-24 18:21:36.286499: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (15)
2018-10-24 18:21:36.486588: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (0)
2018-10-24 18:21:36.686930: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
2018-10-24 18:21:36.887634: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
2018-10-24 18:21:37.087789: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
2018-10-24 18:21:37.288066: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
2018-10-24 18:21:37.487735: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (5)
```

### Further comments:
 - I kind of guessed about the requirements for bridge configuration, before launching 6LBR with make; the `BR-RDC: send failed, slip ack timeout start when the NullPointerException. Is this just bad bridge configuration, or is it related to the simulation stopping?
 - is the NullPointerException related to the earlier errors I noticed after Cooja is launched?
 - is there any more information I can provide to help troubleshoot?
 - is there a newer, more preferred simulation I can run with cooja+6lbr? It's labeled on the wiki as internal and old.

I think, for now, I'll try running things in the InstantContiki3 VM, because obviously that will be easier to support.

laurentderu, 2018-10-26:
The errors in 
https://github.com/cetic/6lbr/files/2512854/cooja-error.txt can safely be ignored, it's just a bug in the Cooja mote compilation that has no consequences (and is solved in Contiki-NG)

The NullPointerException could be due that you are running on a 64bits machine, cooks supports only 32bits architecture right now. The bridge creation has nothing to do with the bridge configuration, it means, as you said correctly, that the simulation has been stopped and so the BR doesn't get replies back from the (simulated) slip-radio.

Instead of using the old test framework, you could instead use the demos found in 6lbr/demo which are more simpler and reliable : https://github.com/cetic/6lbr/tree/develop/examples/6lbr/demo/README.md

zwgraham, 2018-11-01:
I went ahead and bit the bullet installed VMWare and tried the new demos/made a sim of my own, no issues.

Thanks

closed 2018-11-01 by zwgraham