cetic/6lbr #385: can not access [bbbb::100] on ubuntu 12.04 or 18.04

desword, 2018-10-25:
I tried to install the 6lbr on a computer with ubuntu 12.04 (instant contiki 3.0) or 18.04.
I strictly follow the steps on the http://morschi.com/2017/03/28/setting-up-a-6lowpan-border-router-6lbr-using-a-vm-and-slip-radio/, however, I only get the following results:

1. The 6lbr service can not start on ubuntu 12.04 (instant contiki 3.0) when I typing **sudo service 6lbr start**, and using the command **service --status-all** to check. I only get

`[-] 6lbr. `

2. The 6lbr service may be started on ubuntu 18.04, the results of **service --status-all** change to **[+] 6lbr**. However, I can not access [bbbb::100]. Meanwhile, there are no logs at /var/log/6lbr.log, and there is also no file created at /etc/6lbr/nvm.dat.  

On ubuntu 12.04, I have enabled the ipv6 access on firefox browser. and on ubuntu 18.04, I used the chrome which enables the ipv6 automatically.

I only want to setup the 6lbr service on a computer. The slip-radio and other node application have not compiled or tested yet. But I think the latter node application will have no influences on the installation of 6lbr. 

Anyone can tell me how to fix this problem? Thanks very much.

The 6lbr is the latest.

desword, 2018-10-25:
I have also tried on Debian 8.x, However, it doesn't work.

desword, 2018-10-26:
This the 6lbr service status on ubuntu 18.04 computer:

● 6lbr.service - LSB: 6LoWPAN Border Router
   Loaded: loaded (/etc/init.d/6lbr; generated)
   Active: active (exited) since Fri 2018-10-26 15:17:17 +08; 19min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 918 ExecStart=/etc/init.d/6lbr start (code=exited, status=0/SUCCESS)

Oct 26 15:17:17 systemd[1]: Starting LSB: 6LoWPAN Border Router...
Oct 26 15:17:17 systemd[1]: Started LSB: 6LoWPAN Border Router.

desword, 2018-10-26:
However, when I use chrome to access [bbbb::100], it replys with 

```
This site can’t be reached  
http://[bbbb::100]/ is unreachable.  
ERR_ADDRESS_UNREACHABLE  
```

laurentderu, 2018-10-26:
Without a slip-radio connected, 6LBR won't start up. Though you should have a 6lbr.log file with a repeating line saying that it waits for the slip-radio. As yo don't have a 6lbr.log file, this means that you also did not configure /etc/6lbr/6lbr.conf and so even the service startup code can't work.

6LBR is known to work on Ubuntu 14, 16, and 18, though on 18 netstat and ifconfig have been removed so you need a patch that is available in the develop branch.

desword, 2018-10-29:
Now, I have tried to setup the 6lbr on Raspberry 2 B model v1.1.
The final result is that I still can no access the [bbbb::100], but there are 6lbr.log in /var/log/.

**/etc/network/interface**
````
iface eth0 inet static

address 0.0.0.0
auto br0
iface br0 inet dhcp
    bridge_ports eth0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show eth0 | grep ether | awk '{print $2}'`
````

/etc/6lbr/6lbr.conf

```
MODE=ROUTER

RAW_ETH=1

BRIDGE=0

DEV_BRIDGE=br0

DEV_TAP=tap0

DEV_ETH=eth0

RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0

BAUDRATE=115200

LOG_LEVEL=3 #INFO and above only
```

ifconfig

```
br0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fe91:106  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:91:01:06  txqueuelen 1000  (Ethernet)
        RX packets 1893  bytes 260240 (254.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 25  bytes 4192 (4.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether b8:27:eb:91:01:06  txqueuelen 1000  (Ethernet)
        RX packets 1899  bytes 261821 (255.6 KiB)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 30  bytes 4878 (4.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e8:4e:06:5e:45:2b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 47  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

I also compile the slip-radio from the contiki/examples/ipv6/slip-radio on CC2650 launchpad using the following command.

```
cd $CONTIKI_HOME/examples/ipv6/slip-radio
make TARGET=srf06-cc26xx BOARD=launchpad/cc2650 slip-radio.bin
```
Then  I download the slip-radio.bin to the  CC2650 launchpad  using TI Flash programmer. And I connect it to the  Raspberry 2  via /dev/ttyACM0, which I have set in the /etc/6lbr/6lbr.conf

desword, 2018-10-29:
The attached files are the 6lbr logs.

Part of the logs from 6lbr.log

```
2018-10-29 10:45:03.135105: INFO: TAP: Eth MAC address : b8:27:eb:91:01:06
2018-10-29 10:45:03.135447: INFO: ETH: Eth MAC address : b8:27:eb:91:01:06
2018-10-29 10:45:03.135629: INFO: LLSEC: Using 'nullsec' llsec driver
2018-10-29 10:45:03.135863: INFO: 6LBR: Security layer initialized
2018-10-29 10:45:03.136270: ERROR: SLIP: Unknown command
2018-10-29 10:45:03.136567: INFO: HTTP: Starting webserver on port 80
2018-10-29 10:45:03.137264: INFO: NODECFG: Node Config init
2018-10-29 10:45:03.137449: INFO: NODECFG: No node_config.conf file specified
2018-10-29 10:45:03.137572: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:b23:1102
2018-10-29 10:45:03.137875: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:b23:1102
2018-10-29 10:45:03.138370: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2018-10-29 10:45:03.138599: INFO: 6LBR: RA Daemon enabled
2018-10-29 10:45:03.138714: INFO: 6LBR: Checking addresses duplication
2018-10-29 10:45:31.802121: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:b23:1102
2018-10-29 10:45:31.802542: INFO: 6LBR: Starting as RPL ROUTER
2018-10-29 10:45:31.803939: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2018-10-29 10:45:31.804103: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2018-10-29 10:45:31.804227: INFO: UDPS: UDP server started
```

The address bbbb::100 seems setup successfully, However, I still can not access the http://[bbbb::100] on raspberry 2.

The browser is chromium 60.0.3112.89 built on ubuntu 14.04, running on Raspbian 9.4 (32 bit)

[logs_6lbr.tar.gz](https://github.com/cetic/6lbr/files/2523607/logs_6lbr.tar.gz)

desword, 2018-10-29:
for now, I can access [bbbb::100] on windows, which is another computer with the same network address duration.
By setting the following ipv6 address.

ipv6 address: bbbb::101
netmask: 64
ipv6 gateway: bbbb::100

udpasan, 2019-01-05:
Hi also have the same kind of problem. 6lbr log shows it runs finely but I cannot access the webpage through the PC. Could you please share how did u resolve it?

desword, 2019-01-06:
I access the website on windows. And set the Ipv6 address on windows.

udpasan, 2019-01-06:
@desword thanks for the reply. I tried with my laptops in windows as you put in the above comment. But it did not work. Also tried with ubuntu and fedora by making bootable usb sticks but still it did not work. Since Im using raw ethernet mode, I did not make the br0 interface. Is it mandatory ?

desword, 2019-01-06:
Maybe you should first check whether the global address is assigned. By the way, you run the 6lbr on Raspberry or on PC?
You may ref to https://desword.github.io/2018/11/03/cc2650-mqtt-client-example/
But I am sorry that it is the only chinese version.

udpasan, 2019-01-06:
I run 6lbr in the raspberry pi. Did you do it in your PC?

julgil, 2019-12-01:
> for now, I can access [bbbb::100] on windows, which is another computer with the same network address duration.
> By setting the following ipv6 address.
> 
> ```
> ipv6 address: bbbb::101
> netmask: 64
> ipv6 gateway: bbbb::100
> ```
> 
> I can not set the ipv6 address successfully, so I have switched to the windows and it works at least I can see the 6lbr configuration web site.

Followed your step after installing 6lbr....and i perfectly access the webpage from windows....thnx for your help