cetic/6lbr #391: Cannot build 6lbr/Contiki-NG with latest source again.

chenek, 2019-01-11:
I build 6lbr/Contiki-NG with latest source with the following steps:

1. Git 6lbr to home folder
git clone --recursive https://github.com/cetic/6lbr.git
cd 6lbr
git checkout contiki-ng
git submodule update --init

2. Git contiki-ng to home folder
git clone --recursive https://github.com/cetic/contiki-ng.git
cd contiki-ng
git checkout sixlbr
git submodule update --init

3. Change to /6lbr/examples/6lbr and built 6lbr with the following make command
make WITH_CONTIKI=0 CONTIKI=~/contiki-ng WERROR=0 all

However, I see the following error:

`pi@raspberrypi:~/6lbr/examples/6lbr $ make WITH_CONTIKI=0 CONTIKI=~/contiki-ng WERROR=0 all
Makefile:183: CoAP Server is not supported with Contiki-NG
rm -f *.d *.e *.o contiki-ng-native.a *.so obj_native/*.so
rm -rf obj_native
rm -f cetic-6lbr-native.map
rm -f cetic-6lbr.native
for plugin in plugins/dummy plugins/lwm2m-client; do make -C $plugin clean; done
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr/plugins/dummy'
../../Makefile:183: CoAP Server is not supported with Contiki-NG
rm -f *.d *.e *.o contiki-ng-native.a *.so obj_native/*.so
rm -rf obj_native
rm -f cetic-6lbr-native.map
rm -f cetic-6lbr.native
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr/plugins/dummy'
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr/plugins/lwm2m-client'
../../Makefile:183: CoAP Server is not supported with Contiki-NG
rm -f *.d *.e *.o contiki-ng-native.a *.so obj_native/*.so
rm -rf obj_native
rm -f cetic-6lbr-native.map
rm -f cetic-6lbr.native
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr/plugins/lwm2m-client'
rm -rf bin
make TARGET=native clean
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr'
Makefile:183: CoAP Server is not supported with Contiki-NG
rm -f *.d *.e *.o contiki-ng-native.a *.so obj_native/*.so
rm -rf obj_native
rm -f cetic-6lbr-native.map
rm -f cetic-6lbr.native
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr'
make TARGET=native bin/cetic_6lbr_router
make[1]: Entering directory '/home/pi/6lbr/examples/6lbr'
Makefile:183: CoAP Server is not supported with Contiki-NG
make TARGET=native CETIC_6LBR_RPL_ROUTER=1 cetic-6lbr
make[2]: Entering directory '/home/pi/6lbr/examples/6lbr'
Makefile:183: CoAP Server is not supported with Contiki-NG

--------------------------------------------------------------------------------
6LBR Mode: RPL-Router

Multi-Radio: Enabled
RDC:         custom
Framer:      Default
MAC:         custom
LLSEC:       Disabled
Multicast:   Multi-Multicast
RPL:         Multi-RPL
TCP:         Enabled
NAT64:       Enabled

UDP Server:     Enabled
UDP Client:     Disabled
Node Info:      Enabled
Node Config:    Enabled
Webserver:      Enabled
Webserver-Lite: Disabled
CoAP-Server:    Disabled
NVM Proxy:      Disabled
DNS Proxy:      Enabled
--------------------------------------------------------------------------------

mkdir obj_native
  CC        /home/pi/contiki-ng/arch/platform/native/./platform.c
  CC        /home/pi/contiki-ng/arch/platform/native/./clock.c
  CC        /home/pi/contiki-ng/arch/platform/native/dev/xmem.c
  CC        /home/pi/contiki-ng/arch/platform/native/./cfs-posix.c
  CC        /home/pi/contiki-ng/arch/platform/native/./cfs-posix-dir.c
  CC        /home/pi/contiki-ng/arch/platform/native/dev/buttons.c
  CC        /home/pi/contiki-ng/arch/cpu/native/net/tun6-net.c
  CC        /home/pi/contiki-ng/arch/cpu/native/./rtimer-arch.c
  CC        /home/pi/contiki-ng/arch/cpu/native/./watchdog.c
  CC        /home/pi/contiki-ng/arch/cpu/native/dev/eeprom.c
  CC        /home/pi/contiki-ng/arch/cpu/native/./int-master.c
  CC        /home/pi/contiki-ng/arch/cpu/native/dev/gpio-hal-arch.c
  CC        /home/pi/contiki-ng/os/services/slip-cmd/cmd.c
  CC        /home/pi/contiki-ng/os/services/slip-cmd/packetutils.c
  CC        /home/pi/contiki-ng/os/net/mac/csma/csma.c
  CC        /home/pi/contiki-ng/os/net/mac/csma/csma-output.c
  CC        /home/pi/contiki-ng/os/net/mac/nullmac/nullmac.c
  CC        /home/pi/contiki-ng/os/services/ip64/ip64-dns64.c
  CC        /home/pi/contiki-ng/os/services/ip64/ip64-arp.c
/home/pi/contiki-ng/os/services/ip64/ip64-arp.c:97:23: warning: ‘broadcast_ipaddr’ defined but not used [-Wunused-const-variable=]
 static const uint16_t broadcast_ipaddr[2] = {0xffff,0xffff};
                       ^~~~~~~~~~~~~~~~
  CC        /home/pi/contiki-ng/os/services/ip64/ip64-addrmap.c
  CC        /home/pi/contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c
/home/pi/contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c:41:23: **fatal error: 6lbr-ip64.h: No such file or directory
 #include "6lbr-ip64.h"
                       ^
compilation terminated.**
/home/pi/contiki-ng/Makefile.include:325: recipe for target 'obj_native/ip64-ipv4-dhcp.o' failed
make[2]: *** [obj_native/ip64-ipv4-dhcp.o] Error 1
make[2]: Leaving directory '/home/pi/6lbr/examples/6lbr'
Makefile:461: recipe for target 'bin/cetic_6lbr_router' failed
make[1]: *** [bin/cetic_6lbr_router] Error 2
make[1]: Leaving directory '/home/pi/6lbr/examples/6lbr'
Makefile:435: recipe for target 'cetic_6lbr_router-clean' failed
make: *** [cetic_6lbr_router-clean] Error 2
`
How can I fix this to build 6lbr/Contiki-NG successfully?

chenek, 2019-01-14:
I try the following steps to build latest 6lbr/contiki-ng and it builds successfully,
git clone https://github.com/cetic/contiki-ng.git
cd contiki-ng
git submodule update --init

git clone https://github.com/cetic/6lbr.git
cd 6lbr
git submodule update --init

make WITH_CONTIKI=0 CONTIKI=~/contiki-ng WERROR=0 all

However, I cannot access to [bbbb::100] from my Firefox/Chrome browser. I check my 6lbr.log as in the followings and I see no problem. Anyone has any idea about anything wrong in latest 6lbr/contiki-ng?
`Mon 14 Jan 06:51:45 UTC 2019 : Starting 6LBR
[INFO: Main      ] Starting Contiki-NG-release/v4.1-45-gcd3af6c
[INFO: Main      ]  Net: pfe
[INFO: Main      ]  MAC: multi-radio
[INFO: Main      ] Link-layer address 0102.0304.0506.0708
2019-01-14 6:51:45.238809: INFO: SLIP: Allocated slip device 0 -> 0
2019-01-14 6:51:45.241112: INFO: NVM: NVM Magic : 2009
2019-01-14 6:51:45.241449: INFO: NVM: NVM Version : 6
2019-01-14 6:51:45.243266: INFO: SLIP: SLIP started on /dev/ttyACM0
2019-01-14 6:51:45.262506: INFO: BR-RDC: Got MAC 0 : 0012.4b00.0791.1182
2019-01-14 6:51:46.656984: INFO: ETH: RAW Ethernet interface init
2019-01-14 6:51:46.696001: INFO: TAP: opened device eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::a64:a61b:4fbc:25d7/64 scope link
2019-01-14 6:51:46.801235: INFO: 6LBR: Tentative global IPv6 address (WSN) fd00::212:4b00:791:1182
2019-01-14 6:51:46.801652: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
2019-01-14 6:51:46.802008: INFO: 6LBR: RA Daemon enabled
2019-01-14 6:51:46.802277: INFO: 6LBR: Checking addresses duplication
2019-01-14 6:51:47.321817: INFO: SLIP: radio: setting channel: 26
radio: setting pan id to 0xabcd
2019-01-14 6:51:48.803493: INFO: RPL: Configured as DODAG Root fd00::212
:4b00:791:1182
2019-01-14 6:51:48.804204: INFO: 6LBR: Starting as RPL-Router
2019-01-14 6:51:48.810179: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2019-01-14 6:51:48.810884: INFO: HTTP: Starting webserver on port 80
2019-01-14 6:51:48.811560: INFO: NODECFG: Node Config init
2019-01-14 6:51:48.811982: INFO: NODECFG: No node_config.conf file specified
2019-01-14 6:51:48.812421: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf
 : No such file or directory
2019-01-14 6:51:48.813046: INFO: UDPS: UDP server started
2019-01-14 6:51:48.813466: INFO: DNS: DNS proxy started
2019-01-14 6:51:48.813890: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf
 : No such file or directory
2019-01-14 6:51:48.814404: INFO: 6LBR: CETIC 6LBR Started`

jsolderitsch, 2019-03-29:
Hello, I am still getting the error about the missing header file:
```
  CC        /home/pi/contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c
/home/pi/contiki-ng/os/services/ip64/ip64-ipv4-dhcp.c:41:23: fatal error: 6lbr-ip64.h: No such file or directory
 #include "6lbr-ip64.h"
                       ^
compilation terminated.
```

el-dee, 2019-04-02:
That file is not new at all, could you check that your 6lbr repository is on the latest develop branch and the contiki-ng is on the sixlbr branch ?

If so, could yo add the full build log (with the commands to clone, configure the repo and build ?)

laurentderu, 2019-04-02:
That file is not new at all, could you check that your 6lbr repository is on the latest develop branch and the contiki-ng is on the sixlbr branch ?

If so, could yo add the full build log (with the commands to clone, configure the repo and build ?)

Picnotes, 2019-04-16:
hello guys, 
I'm doing the same thing and git failed to find sixlbr folder:

pi@raspberrypi:~/contiki-ng $ git checkout sixlbr
error: pathspec 'sixlbr' did not match any file(s) known to git.

any Idea why?

ahmedmibrahim, 2019-04-20:
I'm facing the same issue:
I used:
    # Download contiki-ng
        cd
        sudo git clone https://github.com/cetic/contiki-ng.git
        cd contiki-ng
        sudo git checkout sixlbr
        sudo git submodule update --init --recursive
    # Download 6lbr
        cd
        git clone https://github.com/cetic/6lbr.git
        cd 6lbr
        git checkout contiki-ng
        git submodule update --init --recursive
    # Install 6lbr with contiki-ng
        cd examples/6lbr
        sudo make WITH_CONTIKI=0 CONTIKI=~/contiki-ng WERROR=0 all
and it showed the same failure.

@laurentderu 
6lbr-ip64.h does not exist in /home/pi/6lbr/examples/6lbr/6lbr if I checked out the repo using the previous commands ...

chenek, 2019-04-23:
I update steps in https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html?m=1 to build 6lbr with Contiki-ng successfully.

closed 2019-04-23 by chenek