cetic/6lbr #409: Connecting CC2530 to 6LBR

newfornow12, 2020-07-27:
Dear all,
I have a Zigbee CC2530 UART Module as it can be seen in this link:

[https://www.openimpulse.com/blog/products-page/product-category/zigbee-cc2530-wireless-module-drf1605h-antenna/]

and I want to connect it to 6LBR running on Raspberry Pi 4 B. I have flashed the module with CC Debugger using a firmware which I made using the following command in the directory /contiki/examples/ipv6/slip-radio :

sudo make TARGET=cc2530dk slip-radio

After that, I connected the module to Raspberry Pi using the UART ports (Tx and Rx) but as the Logfile shows, the 6LBR gets stuck in the "Fetching Mac Address" loop and it does not connect. I would really be glad if anyone helps me with this problem as I got stuck in it for weeks.

My 6lbr.conf file contains the following:
                                                 
```
MODE=ROUTER

RAW_ETH=0
#BRIDGE=0
#DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/serial0
BAUDRATE=115200

LOG_LEVEL=3
```

and my 6lbr.log file shows the following:

```
Tue 28 Jul 2020 12:07:59 AM +0430 : Starting 6LBR
/usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -F /etc/6lbr/factory.dat -o /etc/6lbr/nvm.conf -s /dev/serial0 -R -t tap0 -B 115$
Contiki-contiki-6lbr-1.5.0-2569-gc3092a1cc started with IPV6, RPL
Rime started with address 1.2.3.4.5.6.7.8
MAC multi-radio RDC br-rdc SEC nullsec NETWORK sicslowpan
2020-07-28 0:07:59.763834: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki-contiki-6lbr-1.5.0-2569-gc3092a1cc)
2020-07-28 0:07:59.764032: INFO: SLIP: Allocated slip device 0 -> 0
2020-07-28 0:07:59.764080: INFO: ETH: 6LBR watchdog started (interval: 60)
2020-07-28 0:07:59.764230: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/coap-mqtt-proxy.so
2020-07-28 0:07:59.764587: INFO: PLUGIN: Initialising cmp
2020-07-28 0:07:59.764628: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
2020-07-28 0:07:59.764890: INFO: PLUGIN: Initialising lwm2m-client
2020-07-28 0:07:59.764917: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
2020-07-28 0:07:59.766815: INFO: NVM: NVM Magic : 2009
2020-07-28 0:07:59.766860: INFO: NVM: NVM Version : 6
2020-07-28 0:07:59.766874: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
2020-07-28 0:07:59.766925: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
2020-07-28 0:07:59.766955: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
2020-07-28 0:07:59.766968: INFO: FRAMER: Using 802.15.4 framer
2020-07-28 0:07:59.766980: INFO: MAC: Using 'csma' mac driver
2020-07-28 0:07:59.767228: INFO: SLIP: SLIP started on /dev/serial0
2020-07-28 0:07:59.777416: INFO: SCMD: Started br-cmd process
2020-07-28 0:07:59.777464: INFO: BR-RDC: SLIP RADIO configured as RADIO
2020-07-28 0:07:59.777477: INFO: BR-RDC: Configuring SLIP RADIO 0 (API: 1.0)
2020-07-28 0:07:59.777490: INFO: BR-RDC: Reset SLIP Radio
2020-07-28 0:07:59.777504: INFO: BR-RDC: Fetching MAC address
CMD output: EUnknown command 2152 (2)
2020-07-28 0:08:00.777602: INFO: BR-RDC: Fetching MAC address
2020-07-28 0:08:01.777353: INFO: BR-RDC: Fetching MAC address
2020-07-28 0:08:02.777430: INFO: BR-RDC: Fetching MAC address
```

newfornow12, 2020-07-30:
Any clue guys?

sploitem, 2020-11-11:
Hello.

The device page says:
"Serial Port Rate: 38400 bps (default)".

try this baudrate.